boltgolt/howdy #511: Howdy installed and tests are working but when I use sudo or try to log in camera isn't being used

TimestaticYT, 2021-02-26:
My camera isn't being used and I don't see an error message from Howdy in the terminal.
Camera works as expected and normal when using `sudo howdy test` and `sudo howdy snapshot`

----

Linux distribution: Manjaro Gnome

Howdy version: 2.6.1

sawamoha20, 2021-02-26:
Solved it. From the wiki page "In order for Howdy to authenticate a user, a small change must be added to any PAM configuration file where Howdy might want to be used. The following line must be added to any configuration file:"

You have to add `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/sudo.`

boltgolt, 2021-02-26:
Exactly, it's a more of an involved process on some distros, be sure to fully read the documentation

closed 2021-02-26 by boltgolt