boltgolt/howdy #512: Can't add new camera on Fedora 33

FilipLaurentiu, 2021-03-04:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._
```
[filip@filip-desktop ~]$ sudo howdy add
[sudo] password for filip: 
Adding face model for the user filip
Enter a label for this new model [Initial model] (max 24 characters): Lenovo
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fa2e560b8b0>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```
```
[filip@filip-desktop ~]$ sudo howdy snapshot
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f795d7ba670>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 33

Howdy version (`sudo howdy version`): 2.6.0

boltgolt, 2021-03-04:
Have you configured the correct camera device as the device_path?

FilipLaurentiu, 2021-03-04:
> Have you configured the correct camera device as the device_path?

Than you ! :) I didn't know that I should add my camera to device_path

closed 2021-03-04 by FilipLaurentiu