boltgolt/howdy #513: IR emitters on HP Spectre x360

03l54rd1n3, 2021-03-04:
On my HP Spectre x360 the IR emitters don't turn on when using Howdy on Manjaro KDE.
The video feed when testing howdy is grayscale (so I guess it's in IR mode?) and it does indeed draw a red circle around my face. The only thing not coming on are the two red lights next to the camera lens.

There is only one camera I could use:
```
~ > lsusb            
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 002: ID 064e:3401 Suyin Corp. HP TrueVision FHD RGB-IR
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I searched the available options and also tried all of them through with `v4l2-ctl`. None has an effect on the IR emitters:
```
~ > v4l2-ctl -l -d /dev/video0
brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=100
contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
gamma 0x00980910 (int)    : min=90 max=150 step=1 default=120 value=120
gain 0x00980913 (int)    : min=1 max=7 step=1 default=1 value=1
power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4000 value=4000 flags=inactive
sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=2 value=2
backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=1
exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
exposure_absolute 0x009a0902 (int)    : min=10 max=626 step=1 default=156 value=156 flags=inactive
exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
```
According to some posts I found that the IR camera should be at /dev/video2. I only have 0 & 1 which seems to be the root cause of my problem IMO.
Any help would be appreciated to find the cause of my problem. Thank you :)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Manjaro KDE

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2021-03-05:
It's quite common to only have a video0 and video1, my system has the same.  As for emitters #261 might help you

03l54rd1n3, 2021-03-12:
So I need to debug and dump my usb communication? Gonna be honest here, that's tough, and for me probably a show stopper.

boltgolt, 2021-03-13:
I know that's difficult, and for almost all users it's too technical. There's simply no easy fix for this

EmixamPP, 2021-03-23:
Hello, go have a look on my repo : https://github.com/EmixamPP/linux-enable-ir-emitter.
I think you will find your answer, there everything is explained.
I remain available if you need help :smiley:

03l54rd1n3, 2021-03-23:
Thank you so much! I will try that but it could actually be that I don't have time for this in the next few weeks.
Ich will update here and in your config.txt when I get it (and a darn Windows VM) running.

I will close this issue.

closed 2021-03-23 by 03l54rd1n3

MSloan14, 2026-08-18:
Late, but this is solvable, and I think the greyscale feed is what threw everyone off.

Howdy converts frames to greyscale itself before handing them to dlib. So the test window looks greyscale whether or not the sensor is in IR mode, and the emitters stay dark because the sensor is still in colour. That is what you were seeing.

The sensor does have an IR mode. It is not a separate video node, which is why it looks unreachable.

HP Spectre x360 13-ac0XX, `064e:3401` Suyin "HP TrueVision FHD RGB-IR", Fedora 44, kernel 7.1.8, stock in-tree uvcvideo, no quirks and no udev rules.

`v4l2-ctl --list-formats` shows only MJPG and YUYV on `/dev/video0`, `/dev/video1` is metadata-only, and the descriptors have one VideoStreaming interface with no GREY/Y8 format. The sensor mode-switches instead. Extension Unit 3 carries GUID `96274e36-eca7-49d5-995c-13cf882759cc`, and selector 2 is a one-byte control:

```
1 = colour (default)
2 = infrared, emitters on
```

Write 2 and `/dev/video0` returns greyscale IR frames on the same stream, with the emitters lit. `GET_MIN`/`GET_MAX` report 1 and 3. I have not tested value 3.

How I checked it is really IR and not a desaturated mode: in YUYV the U and V bytes pin to exactly 128, so chroma deviation measures 0.00 rather than merely small, and in a dim room mean luma goes from about 38 to about 91.

It works end to end. Howdy is wired into my KDE lock screen and authenticates in IR. Enrol in IR as well, or the models will not match what the sensor produces at authentication time.

**WARNING.** Writing Extension Unit 4 selector 16 with value 2 hard-hung the camera firmware:

```
uvcvideo 1-5:1.0: Failed to query (SET_CUR) UVC control 16 on unit 4: -110
usb 1-5: device descriptor read/64, error -110
usb usb1-port5: attempt power cycle
usb 1-5: unable to enumerate USB device
```

It stopped answering device descriptor reads. USB reset, port disable and xHCI unbind all failed, and the kernel's own port power cycle did not recover it either. Only suspend to RAM brought it back, presumably because S3 drops power to the module. I have not tried to reproduce it.

So do not point `linux-enable-ir-emitter` at this camera. Its brute-force search sweeps unit 4, and its model assumes a separate greyscale node that does not exist here.

Code, with the safety rails and the measurement: https://github.com/MSloan14/hp-rgbir-ir-mode
It writes exactly one control and refuses the known-dangerous unit-4 selectors by name.

The same unit-3 GUID appears with identical descriptors on at least `064e:3410` and Chicony `04f2:b615`. I have not tested selector 2 on those and cannot say whether it behaves the same.

Disclosure: written with AI assistance and verified on the hardware above. https://github.com/MSloan14/manifesto