boltgolt/howdy #515: Support for multiple IR cameras

larvel, 2021-03-06:
Would it be possible to support two IR cameras? When connected to the external monitor, I prefer not having to open the laptop lid or turn towards the laptop.

When not connected to the external montor, its would be nice to be able to unlock using the laptop IR camera.

tuhlmann, 2021-03-12:
@larvel Have a look at https://github.com/tuhlmann/select-howdy-source

It's a udev rule and a small script that changes the camera for Howdy. You can add multiple cameras and it will choose the first one it finds or disable Howdy.

It's pretty much adapted to my needs, you may need to tinker with the udev rule and of course the camera models.

imswebra, 2021-03-12:
If you want something simpler (and maybe with less features, I haven't looked closely at @tuhlmann's script), you can just use a udev rule to directly change a symlink between your two cameras, and then point Howdy at that symlink. See the post [here](https://github.com/boltgolt/howdy/issues/345#issuecomment-733161807).

scamiran149, 2021-05-30:
I would love to have multiple camera support for multiseat usage.

We have 3 monitors/keyboard/mice setup using loginctl/systemd. Works brilliantly. Would love to have each seat use its own howdy instance with the relevant camera.

Wondering if that would be possible.

Tiger862000, 2022-05-03:
Hey, can you please link some of the solutions in the wiki section! That would be very helpful. Thank you.

FossPrime, 2022-09-29:
This should be a feature request. Many laptops don't operate well with the lid closed, some don't support USB to wake, some are partially or severely cooling restricted, some could get screen damage if GPU heat constantly touches the screen.

I setup Howdy months ago.There is no good place to put my laptop in a way that doesn't obstruct other monitors and leaves line of sight to my face from the built in camera. Currently the builtin camera faces that back of my 50" monitor.

Comma separated values for device_path would fix all this