boltgolt/howdy #517: Downloading 3 required data files... Too SLOW !

forliuyifei, 2021-03-11:
```
sudo dpkg -i howdy_2.6.1.deb
(正在读取数据库 ... 系统当前共安装有 258887 个文件和目录。)
准备解压 howdy_2.6.1.deb  ...
Backup of Howdy config file created in /tmp/howdy_config_backup_v2.6.1.ini
正在解压 howdy (2.6.1) 并覆盖 (2.6.1) ...
正在设置 howdy (2.6.1) ...
Attempting installation of missing data files
Downloading 3 required data files...
```
Then there is noting happened :cry: 

Other part of install howdy seems ok.

forliuyifei, 2021-03-11:
Can you tell me what data files does howdy need?
Maybe there is another way to fetch them for me.

boltgolt, 2021-03-11:
This is the second issue opened by a user with their system language in Chinese where the data files fail to download. Could you give me the output of `wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2`?

forliuyifei, 2021-03-12:
> This is the second issue opened by a user with their system language in Chinese where the data files fail to download. Could you give me the output of `wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2`?

Thx, I will have a try.

aiaiai-czh, 2021-03-14:
> This is the second issue opened by a user with their system language in Chinese where the data files fail to download. Could you give me the output of `wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2`?

hello, i get the dlib_face_recognition_resnet_model_v1.dat.bz2 after i run the command， but when get the false:
bzip2: Can't open input file *.bz2: No such file or directory.
Error while running last command
could you help me?

boltgolt, 2021-03-14:
Are you located in mainland china @aiaiai-czh? Does opening the downloaded archive with your normal archive manager work?

aiaiai-czh, 2021-03-14:
> Are you located in mainland china @aiaiai-czh? Does opening the downloaded archive with your normal archive manager work?

yes, i located in china, and my ubuntu20.04 can't open de downloaded archive, it tips that "TGA image type not supported", how could i solve this problem?

boltgolt, 2021-03-14:
@aiaiai-czh seems like the archive is corrupted. Could you give me exactly what wget prints in the terminal and send me the file that it saves? I'll debug it for you

aiaiai-czh, 2021-03-14:
this file is the wget prints in the terminal

boltgolt, 2021-03-14:
@aiaiai-czh I don't think the file got send, can you upload it somewhere and post the link here?

aiaiai-czh, 2021-03-14:
> @aiaiai-czh I don't think the file got send, can you upload it somewhere and post the link here?

this is the result:

czh@czh-Inspiron-5493:~$ wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
--2021-03-14 17:28:31--  https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
Connecting to 127.0.0.1:58591... connected.
Proxy request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2 [following]
--2021-03-14 17:28:32--  https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2
Connecting to 127.0.0.1:58591... connected.
Proxy request sent, awaiting response... 200 OK
Length: 21428389 (20M) [application/octet-stream]
Saving to: ‘dlib_face_recognition_resnet_model_v1.dat.bz2’

dlib_face_recogniti 100%[===================>]  20.44M  2.45MB/s    in 8.2s    

2021-03-14 17:28:42 (2.50 MB/s) - ‘dlib_face_recognition_resnet_model_v1.dat.bz2’ saved [21428389/21428389]

forliuyifei, 2021-03-16:
> This is the second issue opened by a user with their system language in Chinese where the data files fail to download. Could you give me the output of `wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2`?

```
export http_proxy=127.0.0.1:7890 ; export https_proxy=127.0.0.1:7890
```
*This is my proxy.*
```
wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
--2021-03-16 09:41:31--  https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
正在连接 127.0.0.1:7890... 已连接。
已发出 Proxy 请求，正在等待回应... 302 Found
位置：https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2 [跟随至新的 URL]
--2021-03-16 09:41:32--  https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2
正在连接 127.0.0.1:7890... 已连接。
已发出 Proxy 请求，正在等待回应... 200 OK
长度：21428389 (20M) [application/octet-stream]
正在保存至: “dlib_face_recognition_resnet_model_v1.dat.bz2”

dlib_face_recogni 100%[===========>]  20.44M  6.31MB/s  用时 3.2s    

2021-03-16 09:41:37 (6.31 MB/s) - 已保存 “dlib_face_recognition_resnet_model_v1.dat.bz2” [21428389/21428389])
```
Now I've download it successfully.

But 
```
sudo dpkg -i howdy_2.6.1.deb 
(正在读取数据库 ... 系统当前共安装有 299522 个文件和目录。)
准备解压 howdy_2.6.1.deb  ...
Backup of Howdy config file created in /tmp/howdy_config_backup_v2.6.1.ini
正在解压 howdy (2.6.1) 并覆盖 (2.6.1) ...
正在设置 howdy (2.6.1) ...
Attempting installation of missing data files
Downloading 3 required data files...
```
Nothing goes on still :cry:

aiaiai-czh, 2021-03-16:
> sudo dpkg -i howdy_2.6.1.deb
> (正在读取数据库 ... 系统当前共安装有 299522 个文件和目录。)
> 准备解压 howdy_2.6.1.deb  ...
> Backup of Howdy config file created in /tmp/howdy_config_backup_v2.6.1.ini
> 正在解压 howdy (2.6.1) 并覆盖 [...]

你好，我的ubuntu20.04的系统是英文版的，也没有成功，也是在Downloading 3 required data files...卡着，然后提示bzip2: Can't open input file *.bz2: No such file or directory.

forliuyifei, 2021-03-16:
> 你好，我的ubuntu20.04的系统是英文版的，也没有成功，也是在Downloading 3 required data files...卡着，然后提示bzip2: Can't open input file *.bz2: No such file or directory.

这不是中文英文版的问题, 而是因为所需文件在境外, 不开代理无法连接, 但是我并不知道 dpkg 时如何开启代理, 不知道上面的 " Downloading 3 required data files..." 走的是什么协议, 我明明配置好了 http 、https 代理，ftp 也尝试过，就是无法下载，最好的办法可能是用一部能开代理中继的手机或者路由器实现真正的全局代理，但无奈的是我现在不具备这两种条件.

forliuyifei, 2021-03-18:
我的破手机救回来了，现在已经成功用上 howdy 。

此问题继续保留。

aiaiai-czh, 2021-03-18:
> 我的破手机救回来了，现在已经成功用上 howdy 。
> 
> 
> 
> 此问题继续保留。

你好，能告诉我具体如何操作吗，我开了vpn没用

boltgolt, 2021-03-18:
Please keep your comments in english, i can't help here

aiaiai-czh, 2021-03-18:
> Please keep your comments in english, i can't help here

@forliuyifei say he use his phone make howdy work, i just want to ask him how to do?

aiaiai-czh, 2021-03-18:
i have post the record of the terminal, but i don't konw how to solve this problem

forliuyifei, 2021-03-18:
> Please keep your comments in english, i can't help here

Sorry, I almost gave up before.

I enabled proxy (which helps me connect to the Internet out of China), and used this [VPN Hotspot](https://play.google.com/store/apps/details?id=be.mygod.vpnhotspot) to enable "Repeater" on a rooted phone, then my computer connected to this "hotspot" , and now it can download the required files successfully.

forliuyifei, 2021-03-18:
> > 我的破手机救回来了，现在已经成功用上 howdy 。
> > 此问题继续保留。
> 
> 你好，能告诉我具体如何操作吗，我开了vpn没用

1. A rooted Android phone.
2. A proxy tool that works fine on your phone.
3. Download and install VPN Hotspot from Google Play.
4. Make sure that your phone connect to a Wi-Fi.
5. Enable both the proxy & Repeater
[image]
6. Let your computer connect the hotspot which is created by  your phone.(It not same as the system one)
7. `sudo dpkg -r howdy`
8. `sudo dpkg -i howdy*.deb` , wait and go on.

aiaiai-czh, 2021-03-18:
> Sorry, I almost gave up before.
>
>
>
> I enabled proxy (which helps me connect to the Internet out of China), and used this [VPN [...]

我直接电脑连接vpn下载不可以吗

forliuyifei, 2021-03-18:
> 我直接电脑连接vpn下载不可以吗

反正我没有找到真正的全局代理方法😓

aiaiai-czh, 2021-03-18:
> 反正我没有找到真正的全局代理方法😓

好的，多谢了🙏

boltgolt, 2021-03-29:
Thanks all for figuring it out, was unfortunately not much i could do from here. Closing this now because the problem is not with Howdy

closed 2021-03-29 by boltgolt

Syize, 2021-04-22:
1. cd into your howdy lib directory, which should in "/lib/security/howdy/dlib-data". You can find it in /var/lib/dpkg/info/howdy.postinst, written in around line 91 ( to locate it I insert some code so I don't know its original location)
2. modify install.sh, I use proxychains4 to enable wget( I have wget installed ) to use my proxy
3. sudo apt install again to finish installation
4. enjoy it

Syize, 2021-04-22:
On my Ubuntu, I use method below to install howdy when data downloading failed.

1. cd into your howdy library directory, which should in **/lib/security/howdy/dlib-data**
2. modify **install.sh**, I use proxychains4 to enable wget to use my proxy
3. cd into dpkg library, which in **/var/lib/dpkg/info**, and run
`sudo ./howdy.preinst install`
to generate **howdy_picked_device** to make dpkg think it's fresh configuration mode
4. modify **line 131** in **howdy.postinst**, change it from
`LOADER_CMD = [loader, "--tries", "5", "--output-document"]`
to
`LOADER_CMD = ["proxychains4", "-q", loader, "--tries", "5", "--output-document"]`
so that wget can use proxy to download file
5. finally you can run
`sudo apt install howdy`
or whatever command that can make apt continue installing howdy

**Here is some advise**
1. You can specify which python3 packages howdy needs in README so people can install them before, in case root's pip use different pypi mirror.
2. You can keep **howdy_picked_device** file existing until installation finally complete, in case installation failed. It took me a while to find why the script can't run properly again after I modified it because the file was deleted.

aiaiai-czh, 2021-04-22:
Thanks for your advice! I will try it later.

    

            On 04/22/2021 23:59, Syize wrote: 
On my Ubuntu, I use method below to install howdy when data downloading failed.

cd into your howdy library directory, which should in /lib/security/howdy/dlib-data
modify install.sh, I use proxychains4 to enable wget to use my proxy
cd into dpkg library, which in /var/lib/dpkg/info, and run
sudo ./howdy.preinst install
to generate howdy_picked_device to make dpkg think it's fresh configuration mode
modify line 131 in howdy.postinst, change it from
LOADER_CMD = [loader, "--tries", "5", "--output-document"]
to
LOADER_CMD = ["proxychains4", "-q", loader, "--tries", "5", "--output-document"]
so that wget can use proxy to download file
finally you can run
sudo apt install howdy
or whatever command that can make apt continue installing howdy

Here is some advise

You can specify which python3 packages howdy needs in README so people can install them before, in case root's pip use different pypi mirror.
You can keep howdy_picked_device file existing until installation finally complete, in case installation failed. It took me a while to find why the script can't run properly again after I modified it because the file was deleted.

[
{
"target": "https://github.com/boltgolt/howdy/issues/517#issuecomment-824969477",
"url": "https://github.com/boltgolt/howdy/issues/517#issuecomment-824969477",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

SpinningK, 2022-03-21:
You may lose your proxy env when you use sudo, which makes the download too slow. To solve that, you can first switch to root with
``` shell
su -s
```
Then export proxy env like
``` shell
export http_proxy=xxxxxxxxxxx
export https_proxy=xxxxxxxxxxx
```
Finally install howdy right in the root shell with
```shell
apt install howdy
```

Solarweapon, 2022-06-10:
> You may lose your proxy env when you use sudo, which makes the download too slow. To solve that, you can first switch to root with
> 
> ```shell
> sudo -s
> ```
> 
> Then export proxy env like
> 
> ```shell
> export http_proxy=xxxxxxxxxxx
> export https_proxy=xxxxxxxxxxx
> ```
> 
> Finally install howdy right in the root shell with
> 
> ```shell
> apt install howdy
> ```

Thank you for your invaluable help!

chenxiex, 2022-08-06:
> You may lose your proxy env when you use sudo, which makes the download too slow. To solve that, you can first switch to root with
>
> ```shell
> sudo -s
> ```
>
> Then export proxy env like
>
> ```shell
> export [...]

I don't know why but this doesn't help. It does establish a connection through proxy, but github blocks the connection out. This method is of great help when downloading the pip package anyway. In my case, the only way to download the 3 files is through proxychains.