boltgolt/howdy #518: Open a dialog or perform other action if authentication is being performed

tuhlmann, 2021-03-12:
I sometimes do not recognize that Howdy is trying to authenticate me and do not look into the camera, so login fails.

It would be nice if there was a way to tell Howdy to do something while it's scanning, like opening a window just saying "Scan in Progress".

I read the Wiki about the rubber stamp guide which sounded like that maybe an extension point that I could use, but I didn't find any examples or further documentation.

Is this rubber stamp mechanism still existing?

boltgolt, 2021-03-12:
Good news, this has already been made and will be in the 3.0.0 release! The rubber stamp guide is being written in preparation for that release, they are ready to go. 

Unfortunately a big part of this release is not done yet. Using a native PAM module is a breaking change and I want to combine it into the 3.0.0 version. There's no launch date yet

tuhlmann, 2021-03-12:
Thanks for the quick answer!

I found this flag in config: `detection_notice = true`, which does what I need. It shows a string in the command line and a dialog when a gui app tries to authenticate!

I'm still very curious to learn what I could do with rubber stamps. Looking forward!

Thanks!

closed 2021-03-12 by tuhlmann