boltgolt/howdy #519: Keeping snapshots should be disabled by default

Juerd, 2021-03-14:
So I ran `howdy config`, and at the bottom encountered some settings regarding snapshots. Went looking and found a lot of images of my face, but also some unintended authentication attempts featuring other body parts. (xsecurelock attempts authentication when the mouse cursor is moved, which can happen without any intention to actually use the computer.) Was not amused to find these images on my drive, and that they're mode 644 doesn't really help either (#397), to put it mildly.

I'm not sure why I hadn't noticed this functionality before, but I think the default setting should be disabled because it doesn't have any role in the primary feature of the software (authentication, not taking pictures), while it does add some serious risks. And if it is kept as a default setting anyway, it should come with some very clear warnings regarding the possible consequences of taking pictures.

boltgolt, 2021-03-14:
It really should not be, and will be [disabled automatically](https://www.github.com/boltgolt/howdy/tree/dev/src%2Fconfig.ini) in the next release. Unfortulately I left these config values on true while testing and I did not catch this error when I was checking the release version

boltgolt, 2021-03-14:
Doe debian systems there's also a [check](https://www.github.com/boltgolt/howdy/tree/dev/debian%2Fpostinst#L80) on upgrade that disables snapshot and notifies the user

Juerd, 2021-03-14:
Thanks for the quick reply. I'm glad it's a known problem and fixed already. Please be careful with future releases.

closed 2021-03-14 by Juerd