boltgolt/howdy #521: Word on Security

Amarjit3, 2021-03-18:
In the README, it says "well-printed photo of you could be enough to do it [spoof Howdy]". Could someone explain how this is the case? If an IR capture method is being used to authenticate the person, then surely only a photograph taken with an IR camera could in theory spoof Howdy?

boltgolt, 2021-03-18:
If you print a color photo with some normal inks it will also show up in the IR spectrum. For example, on an old Swedish map i had laying around only the black (K) ink was visible in IR while the colors (CMY) do not absorb any IR light. If you print your face in black and white with the black ink it would be indistinguishable to your actual face for the camera.

Visible light on the left, IR on the right:
[image]

Amarjit3, 2021-03-18:
Oh that's really interesting - thanks for the explanation! I suppose then Windows Hello is also susceptible to this?

boltgolt, 2021-03-18:
Windows Hello uses the two IR emitters to generate a 3D image of your face, and is much more secure. To do this it lights up your face with the left emitter on even framed, and uses the right emitter on odd frames. This lights up your face in slightly different angles, which is not possible to be faked by simply printing a 2D photo.

Unfortunately Howdy does not have control over these IR emitters and can't use this process

Amarjit3, 2021-03-18:
Perfect, thanks for the details.

closed 2021-03-18 by Amarjit3