boltgolt/howdy #522: Ubuntu 20.04 PAM problem

overlordhu, 2021-03-20:
Hi,

I recently installed Howdy from binary packages from PPA on Ubuntu 20. It is a freshly installed OS, almost empty, recently set up email and other system accounts, and have not used it for more than a few days now, so it can be considered clean.

The issue I am seeing is that howdy works great in terms of GDM and sudo in a terminal window for example, but as soon as I enable it and finish my OS session (be it logout-login or reboot) and start a new session there are two specific applications that cease to work from then on:

- element-desktop
- evolution

I do not know what these have to do with PAM or any Linux authentication that can be affected by Howdy, but the problem instantly disappears after i issue 'sudo howdy disable 1', no need to reboot.

I did not recognize any other apps having problems, otherwise Howdy works great. Therefore I do not think it is application version specific, 

Versions:
OS: Ubuntu 20.04.2 LTS
Howdy 2.6.1
element-desktop 1.7.23
evolution 3.36.4

Best,

Pal

boltgolt, 2021-03-22:
This seems like an issue in PAM, not sure what's going wrong exactly but it sounds like it's failing to start Howdy. I think the easiest solution would be to [manually enable](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types) Howdy in the places you want to enable it in.

overlordhu, 2021-03-22:
I disabled howdy module in pam-auth-update as per your manual link writes above.

After that I inserted the line to /etc/pam.d/sudo and I can tell sudo works great and there are no application problems during the session.

My original /etc/pam.d/sudo looked like this:

`%PAM-1.0

session    required   pam_env.so readenv=1 user_readenv=0
session    required   pam_env.so readenv=1 envfile=/etc/default/locale user_readenv=0

@include common-auth
@include common-account
@include common-session-noninteractive`

I inserted howdy module statement right before the first @include, so now it looks like this:

`%PAM-1.0

session    required   pam_env.so readenv=1 user_readenv=0
session    required   pam_env.so readenv=1 envfile=/etc/default/locale user_readenv=0

auth       sufficient   pam_python.so  /lib/security/howdy/pam.py

@include common-auth
@include common-account
@include common-session-noninteractive`

I also played around with /etc/pam.d/gdm-password but the application problems reappeared after setting it and rebooting the machine each and every time.
I'm going to play on with pam and get back to you in case I succeed somehow.

rubemalmeida, 2023-10-27:
> Eu desativei o módulo howdy no pam-auth-update conforme seu link manual escreve acima.
> 
> Depois disso eu inseri a linha para /etc/pam.d/sudo e posso dizer que o sudo funciona muito bem e não há problemas de aplicação durante a sessão.
> 
> Meu original /etc/pam.d/sudo ficou assim:
> 
> '%PAM-1,0
> 
> sessão necessária pam_env.so readenv=1 user_readenv=0 sessão necessária pam_env.so readenv=1 envfile=/etc/default/locale user_readenv=0
> 
> common-auth common-account common-session-noninteractive'
> 
> Eu inseri a instrução do módulo howdy logo antes do primeiro, então agora fica assim:
> 
> '%PAM-1,0
> 
> sessão necessária pam_env.so readenv=1 user_readenv=0 sessão necessária pam_env.so readenv=1 envfile=/etc/default/locale user_readenv=0
> 
> auth suficiente pam_python.so /lib/security/howdy/pam.py
> 
> common-auth common-account common-session-noninteractive'
> 
> Eu também brinquei com /etc/pam.d/gdm-password, mas os problemas do aplicativo reapareceram depois de configurá-lo e reiniciar a máquina toda vez. Vou brincar com a pam e voltar para você caso eu consiga de alguma forma.

How did you solve it?

rubemalmeida, 2023-10-27:
> I disabled howdy module in pam-auth-update as per your manual link writes above.
>
> After that I inserted the line to /etc/pam.d/sudo and I can tell sudo works great and there are no application [...]

How did you solve it?

overlordhu, 2023-10-27:
I use a different machine to my day-to-day work since then, but as far as I remember I still have that install in its 20.04 state, so I can hopefully check it out in the next few days for you.

closed 2026-11-16 by boltgolt