boltgolt/howdy #526: Recommended Install Script

JJGadgets, 2021-03-29:
I am looking into creating a Void Linux xbps-src template to install Howdy, but when looking at the .deb post install script and the AUR’s PKGBUILD, I find that they are different, don’t remember off the top of my head but while sharing similarities, one was longer than the other. Which should I refer to?

boltgolt, 2021-03-30:
The arch version has everything needed, but the debian scripts also do extra stuff, like automatically detecting the camera

JJGadgets, 2021-04-01:
Alright, thanks. Will leave this open in case of any other questions while making the xbps-src template.

pandom79, 2021-05-15:
> Alright, thanks. Will leave this open in case of any other questions while making the xbps-src template.

I have a xbps-src template which works. I can't understand the reason often the camera is stucked with led turned ON...;)