boltgolt/howdy #527: Any face is a valid face

avzuquete, 2021-04-02:
Hi!

I've installed and experimented Howdy and it simply recognizes if it is a face.
I've set it to use CNN instead of HOG, certainty=10, but the result is always the same: if a face is presented, any face, live or in a picture, it accepts it as the correct face. Male, female, etc., doesn't mater.

What am I doing wrong? Is this the expected behavior?

Thank you!
----
I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Bundgie, Ubuntu 20.10
Linux version 5.8.0-48-generic (buildd@lgw01-amd64-052) (gcc (Ubuntu 10.2.0-13ubuntu1) 10.2.0, GNU ld (GNU Binutils for Ubuntu) 2.35.1) #54-Ubuntu SMP Fri Mar 19 14:25:20 UTC 2021

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2021-04-03:
Makes sense that certainty 10 is matching any face, 1 is the most certain and 7 the least. I'd recommend setting it to 3 or lower

avzuquete, 2021-04-03:
Oops, sorry, got it wrong!

I though 10 was the most accurate!

Tried the default value of 2.8 and worked fine.

With 2 I only got one match in about 20 attempts.

With 1 always failed ...

CNN all the time.

Thank you!

Regards,

boltgolt, 2021-04-07:
No problem, it's a bit counter-intuitive this way but that's how the underlying library implements this value

closed 2021-04-07 by boltgolt