boltgolt/howdy #530: Removing software while camera is in use causes an infinite loop

4a454646, 2021-04-09:
1. Have your camera already in use from an application like Zoom or Discord.
2. Perform an action that brings up Ubuntu's GUI authentication, such as trying to remove software.
This results in an infinite loop of authentication with no opportunity to enter your password (see attached video). 
Linux distro: Ubuntu 20.04 LTS
Howdy version: Howdy 2.6.1

Video:
https://user-images.githubusercontent.com/46868596/114211526-9b12de00-9915-11eb-8e41-f64eade24d94.mp4

boltgolt, 2021-04-13:
Is howdy your only authentication option? Does this also show up in other places like sudo when your camera is used by zoom/discord?

4a454646, 2021-04-13:
Yes, I have Howdy as the main authenticator, and if it fails I just use the normal password enter. The only places this occurs is when the GUI authentication field is brought up (removing software and trying to perform file operations as admin). There are probably other things that cause the GUI to pop up, but I haven't encountered them yet.

boltgolt, 2021-04-13:
This sounds more like a PAM issue with your GUI implementation. My advice would be to follow [this guide](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types) to only use Howdy where PAM is not buggy

closed 2021-04-21 by 4a454646