boltgolt/howdy #533: Camera path is not configured correctly

MoRmdn, 2021-04-14:
Enter a label for this new model [Initial model] (max 24 characters): MoRmdn
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f3fd21e55e0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): ubuntu

Howdy version (`Howdy 2.6.1
`):

boltgolt, 2021-04-14:
Have you configured your device?

MoRmdn, 2021-04-14:
> Have you configured your device?
> 
> Quote from your error:
> ``` Camera path is not configured correctly, please edit the 'device_path' config value.```

Now i did .. and work perfectly thnx🙏

closed 2021-04-14 by boltgolt

askielboe, 2021-04-29:
If anyone wonders how to edit the `device_path`:

`ls -l /dev/v4l/by-path/` to find your camera. In my case it was `/dev/v4l/by-path/pci-0000:00:14.0-usb-0:7:1.2-video-index0`.

Then edit your howdy config using `sudo howdy config` and assign your video device path to `device_path`.

After that `sudo howdy add` worked fine for me.

c0tt0nm0uffxx, 2023-07-13:
> If anyone wonders how to edit the `device_path`:
>
> `ls -l /dev/v4l/by-path/` to find your camera. In my case it was `/dev/v4l/by-path/pci-0000:00:14.0-usb-0:7:1.2-video-index0`.
>
> Then edit your howdy [...]

Thanks!