boltgolt/howdy #534: Doesn't work in Opensuse Tumbleweed

lixiangwuxian, 2021-04-16:
I followed the guide on [opensuse wiki](https://en.opensuse.org/SDB:Facial_authentication) and installed the howdy. Then I added a model. However, when I try `sudo howdy test`, I got the error

```
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/local/lib64/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

[1]    311 IOT instruction  sudo howdy test
```
I am using KDE on x11.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Opensuse Tumbleweed 20210408

Howdy version (`sudo howdy version`):Howdy 2.6.1

boltgolt, 2021-04-17:
Seems like a Qt problem to me, not necessarily Howdy

lixiangwuxian, 2021-04-19:
I found an interesting way to make howdy works. Run `nano /usr/bin/howdy` then comment out the script about sudo and root detection, and just run `howdy test`, the window will display as normal. I found the reason why `sudo howdy test` not working is X server did not give sudo a display. Not pretty sure if it would work when login. I will try later.

lixiangwuxian, 2021-04-19:
Okay, howdy is fully functional now. Thanks a lot.

closed 2021-04-19 by lixiangwuxian

soltanix, 2022-03-18:
This is because root is not in the list of users that can connect to X server.
Why does Howdy require root privileges for `howdy test`?

boltgolt, 2022-03-19:
Because in a secure installations your models are only readable by root

mscratch, 2022-11-13:
Yes, it's the sudo's problem, openSUSE restricted root's access for GUI, [a guide of SLE](https://documentation.suse.com/sles/12-SP4/html/SLES-all/cha-adm-sudo.html) should address the problem:

> To use X.Org in applications started with sudo, the environment variables DISPLAY and XAUTHORITY need to be propagated. To configure this, create the file /etc/sudoers.d/xorg, (see [Section 2.2.1, “Editing the Configuration Files”](https://documentation.suse.com/sles/12-SP4/html/SLES-all/cha-adm-sudo.html#sec-sudo-conf-edit)) and add the following line:

> `Defaults env_keep += "DISPLAY XAUTHORITY"`

rokejulianlockhart, 2023-07-15:
@hartlottery, couldn't the Howdy installation process account for this and modify the file as needed?

hartlottery, 2023-07-16:
> @hartlottery, couldn't the Howdy installation process account for this and modify the file as needed?

Maybe the operations to the sudo file are quite dangerous, in security manner, I think howdy may just leaving it as a manual operation 🤔.