boltgolt/howdy #535: Completely switch to python3

ava1ar, 2021-04-17:
pam-python 1.0.8 can be build using python3. At this moment, howdy has a workaround applied do support python2 for pam-python (like [here](https://github.com/boltgolt/howdy/blob/master/src/pam.py#L10)). Now we can switch to pam_python 1.0.8 and completely drop Python2-specific stuff.

boltgolt, 2021-04-17:
Good idea, but we're taking it a step further and rewriting the PAM part in c++, checkout the WIP PR

closed 2021-04-18 by boltgolt

ashkop, 2023-04-21:
I've just upgraded to Ubuntu 23.04 and it seems that this prevents howdy from running. That's what I get in the `auth.log`:
```
2023-04-21T10:38:50.406977+03:00 alex-ZenBook /lib/security/howdy/pam.py[14914]: Traceback (most recent call last):
2023-04-21T10:38:50.407061+03:00 alex-ZenBook /lib/security/howdy/pam.py[14914]:   File "/lib/security/howdy/pam.py", line 10, in <module>
2023-04-21T10:38:50.407095+03:00 alex-ZenBook /lib/security/howdy/pam.py[14914]:     import ConfigParser
2023-04-21T10:38:50.407130+03:00 alex-ZenBook /lib/security/howdy/pam.py[14914]: ModuleNotFoundError: No module named 'ConfigParser'
```