boltgolt/howdy #536: Errors during install: "Py_DEPRECATED" and "Usually an int but a long on Cygwin64"

PJSingh5000, 2021-04-17:
On Ubuntu 20.04, I have added the howdy PPA: `ppa:boltgolt/howdy`.

I install howdy as follows:

    export HOWDY_NO_PROMPT=true
    apt install howdy

I see the following errors repeated over 100 times during installation.
Howdy installs fine, and works, but I thought I'd share this in case something can be improved or fixed.

    In file included from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/cast.h:16,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/attr.h:13,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:43,
                     from /tmp/dlib-19.16/dlib/../dlib/python/pybind_utils.h:6,
                     from /tmp/dlib-19.16/dlib/../dlib/python.h:6,
                     from /tmp/dlib-19.16/tools/python/src/opaque_types.h:6,
                     from /tmp/dlib-19.16/tools/python/src/dlib.cpp:4:
    /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/detail/internals.h:82:34: warning: ‘int PyThread_create_key()’ is deprecated [-Wdeprecated-declarations]
       82 |     decltype(PyThread_create_key()) tstate = 0; // Usually an int but a long on Cygwin64 with Python 3.x
          |                                  ^
    In file included from /usr/include/python3.8/pystate.h:10,
                     from /usr/include/python3.8/genobject.h:11,
                     from /usr/include/python3.8/Python.h:121,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/detail/common.h:111,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pytypes.h:12,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/cast.h:13,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/attr.h:13,
                     from /tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:43,
                     from /tmp/dlib-19.16/dlib/../dlib/python/pybind_utils.h:6,
                     from /tmp/dlib-19.16/dlib/../dlib/python.h:6,
                     from /tmp/dlib-19.16/tools/python/src/opaque_types.h:6,
                     from /tmp/dlib-19.16/tools/python/src/dlib.cpp:4:
    /usr/include/python3.8/pythread.h:100:36: note: declared here
      100 | Py_DEPRECATED(3.7) PyAPI_FUNC(int) PyThread_create_key(void);
          |                                    ^~~~~~~~~~~~~~~~~~~

----
I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): 
Ubuntu 20.04

Howdy version (`sudo howdy version`):
Howdy 2.6.1

Python 3 version:
3.8.5

ConNero20, 2021-05-04:
Yep, same (or at least similar) for me on Ubuntu budgie. 

Though I wonder what is the issue/root cause here.

arifer612, 2021-06-16:
The issue here is that the version of `dlib` installed by howdy is 19.6, which is quite old compared to the latest 19.22 release which solves the exact issue you brought up here, as reflected in [this commit](https://github.com/davisking/dlib/commit/2ef8e3ac14671cf8a35690840f3e595f8121ddce). 

I agree that it would be a safer bet to future-proof possible breaks when newer versions of Python comes out and maybe try it out on the dev branch to see if the latest version of `dlib` installs howdy just fine.

Just something on my part, I have verified it to install without any issue on elementary OS Hera, which is based off Ubuntu 18.04

rjousse18, 2021-11-15:
is there a solution? because I have the same error and after installation I cannot launch howdy

closed 2026-11-16 by boltgolt