boltgolt/howdy #539: Can't access ir camera on Razer Blade 15

lagmoellertim, 2021-04-22:
I am able to access the normal (non-ir) camera (it's on /dev/video0). When I try to use the camera with the ir blaster (/dev/video2), I get the following error:
```
$ sudo howdy add
Adding face model for the user [username]
Enter a label for this new model [Model] (max 24 characters): 
[ WARN:0] global /tmp/pip-req-build-ms668fyv/opencv/modules/videoio/src/cap_v4l.cpp (1004) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.

Please look straight into the camera
[ WARN:0] global /tmp/pip-req-build-ms668fyv/opencv/modules/videoio/src/cap_v4l.cpp (1004) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.
Failed to read camera specified in your 'device_path', aborting
```
This is the list of devices
```
$ v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:00:14.0-7):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
```

And here are the supported formats for each of them

```
$ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
	[1]: 'YUYV' (YUYV 4:2:2)
```

```
$ v4l2-ctl -d /dev/video1 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
```
$ v4l2-ctl -d /dev/video2 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
```
```
$ v4l2-ctl -d /dev/video3 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
----

Linux distribution: Ubuntu 20.04 LTS

Howdy version: 2.6.1

closed 2021-05-03 by lagmoellertim

himekifee, 2022-08-06:
How did you sort it out? Having similar issue.

lagmoellertim, 2022-08-06:
I think I needed to configure howdy to request a square image from the camera (so a 1:1 format instead of say 3:4 or 16:9).

himekifee, 2022-08-06:
It works. Thanks.

BloodDrag0n, 2022-10-20:
So, does the IR camera detection working now?

mathix420, 2024-03-05:
I had the same issue on Razer Blade Stealth 13, I used OBS to find which resolution works with the IR camera, and then filled the values in the config file. Thanks @lagmoellertim!

```ini
[video]
# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = 360
frame_height = 360
```