boltgolt/howdy #541: Wrong configuration for VideoCapture in config, picture is too dark

Syize, 2021-04-23:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

----

The device configuration "device_path" in config is wrong.
When I use the configuration in config.ini, howdy can't recognize me because the picture it took is too dark
Here is the info when I use howdy
[image]
And here is what howdy takes
[image]
When I change the code to **VideoCapture(0)**, it works correctly

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu with KDE, 20.04.1 LTS

Howdy version (`sudo howdy version`): 2.6.1

ConNero20, 2021-05-04:
What do you see when you run `sudo howdy test`?

In the case you do not have video output, could you try what's describes [here](https://github.com/boltgolt/howdy/issues/542#issuecomment-829592805) and tell us if it works?

boltgolt, 2021-05-17:
Strange, `VideoCapture(0)` should just make it point to `/dev/video0`, so having that as your device path should do the exact same