boltgolt/howdy #542: sudo howdy add error

ALENTL, 2021-04-25:
AttributeError: 'VideoCapture' object has no attribute 'internal'

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (pop-os Ubuntu based):

Howdy version (`sudo howdy version`): latest release as of 25.4.2021

ConNero20, 2021-04-29:
I had a similar issue and I resolved it the following way:

1. To check if my camera was detected by my OS, I installed v4l2-utils with `sudo apt-get install v4l-utils` I got this from [here](https://askubuntu.com/questions/348838/how-to-check-available-webcams-from-the-command-line#848390).
2. Then, I ran` v4l2-ctl --list-devices`. This is the output for me:

```
Chicony USB2.0 Camera: Chicony  (usb-0000:09:00.3-4):
	/dev/video0
	/dev/video1
```

2. Then, I edited the howdy configuration file under /usr/lib/security/howdy/config.ini. In line 41, for `device_path =` I inserted `/dev/video0` which is one of the lines of my v4l2-ctl output from before.
3. I tested the configuration by running `sudo howdy test` after a timeout and typing my password, the expected test window with b/w (probably infrared?) image opened. I could proceed using howdy now.

I hope this resolves the issue for you as well.

louis993546, 2021-05-03:
Tried it on elementaryOS + T480, other than having to set the `device_path` to `/dev/video2/` which is the normal camera, everything is now working as expected :+1: 

Thanks @ConNero20!

ALENTL, 2021-05-03:
Thanks for the help

closed 2021-05-03 by ALENTL

ConNero20, 2021-05-03:
By the way, this might be a possible duplicate of issue #466

yonikremer, 2023-02-13:
@ConNero20 Thanks! It solved my issue. Can someone please add this to the README or docs of the project