boltgolt/howdy #543: IR emitters not working on Dell Inspiron 5567

winux1, 2021-05-08:
On my dell Inspiron 5567 laptop the IR emitters not turning on (no red leds) when I try to use howdy. The image grayscale and my face is detected. This is the only webcam in my system but I have three video devices: `/dev/video0`, `/dev/video1` and `/dev/media0`. Only `/dev/video0` working with howdy.

There is only one camera I could use according to lsusb:
````
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 0cf3:e005 Qualcomm Atheros Communications 
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 0bda:58c2 Realtek Semiconductor Corp. Integrated Webcam HD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
````
The output of v4l2-ctl -l -d /dev/video0
````
brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=3 value=3
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=10 max=625 step=1 default=166 value=166 flags=inactive
````
For v4l2-ctl -l -d /dev/video1 I get nothing

I tried everything including https://github.com/EmixamPP/linux-enable-ir-emitter but without success.
I get this output when trying to run sudo  python auto-config.py:
````
Compilation of the C script ...
Trying all know infrared camera configuration ...
Configuration #0 ...
Unable to open a file descriptor for /dev/video2
Configuration #0 does not work
Configuration #1 ...
Unable to open a file descriptor for /dev/video2
Configuration #1 does not work
Configuration #2 ...
Unable to open a file descriptor for /dev/video2
Configuration #2 does not work
Configuration #3 ...
Unable to open a file descriptor for /dev/video2
Configuration #3 does not work
````
By the way I don't have a  /dev/video2 device
Thanks for your help in advance!

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux

Howdy version (`sudo howdy version`): Howdy 2.6.1

EmixamPP, 2021-05-09:
I'm linking the issue you opened on my repo, in case it helps https://github.com/EmixamPP/linux-enable-ir-emitter/issues/9.

jaimierodricks3, 2021-05-11:
same issue on Ubuntu on update 
dlib has to be installed using python 3 
but tried many methods after hours it shows error saying packages incomplete

EmixamPP, 2021-05-11:
> but tried many methods after hours it shows error saying packages incomplete
Same issue ?

BackMountainDevil, 2021-05-28:
does this really need an IR emitter? For windows hello, it really that thing because I see it turn red. But for howdy on arch, it only  open the camer and the camera light turn white.  
device: Lenovo Xiaoxin Pro13

EmixamPP, 2021-05-28:
@BackMountainDevil So if it doesn't turn red it doesn't work properly. The purpose of the infrared emitter is to highlight you. So that you can always be identified. Example: put yourself in the dark and try with Howdy to log in, it won't see you. With Windows it will work because the infrared emitter will have revealed you.

Under Linux, this does not always work, because the Kernel does not natively support all the features of your camera. However, with a little research you can get the infrared part of the camera to work in most cases.

If you're interested in making it work, check out my repo:  https://github.com/EmixamPP/linux-enable-ir-emitter

BackMountainDevil, 2021-06-01:
@EmixamPP What impress me is that I search 'IR emitters' in arch wiki. Then I see [this](https://wiki.archlinux.org/title/Dell_XPS_13_(9310))  
>When the IR camera(/dev/video2) is on, it will not automatically turn on the IR emitter. You can follow the instructions from [linux-enable-ir-emitter](https://github.com/EmixamPP/linux-enable-ir-emitter) to enable the IR emitter. 

And I click it and it turns out your repo

BackMountainDevil, 2021-06-01:
Thanks to your reply. I make a right config of my camera. I used to use /dev/video0. Now with /dev/video2 the IR emitter does work. However, the fram it got is smaller.
- /dev/video0:  only white light. 1280x720
- /dev/video2: Both white light and red light. 640x360

I use both dev to take test face under light and dark. Form the photos, I think the /dev/video0 look better.  If the video0 could take bigger frame, it will work better. But I use `v4l2-ctl -d /dev/video2 --list-formats-ext` to check it could not.

Moreover, have you notice the windows hello use the IR emitter, it will blink. But I use howdy with IR emitter, it just keeps red without blink.

EmixamPP, 2021-06-01:
@BackMountainDevil on my computer the emitter blink. And the wight light you're talking about is just the camera usage indicator. 
I think you need a little more help with your infrared camera setup. Since the problem is not the same as this issue, I can suggest you to either open another one on this repo or one on mine (feel free to tag me). 

Just in case: did you run the python script `auto-config.py` I mentioned in my readme? If not, I advise you to do so, before opening an other issue.