boltgolt/howdy #544: Unknown Error: 1 - Only in Log Screen / sudo working - Fedora 34

gabrieleaquaro, 2021-05-13:
Hello everyone, 

I'm running Howdy on Fedora 34 (gnome 40)  and i'm having some troubles.
First of all the installation is fine and everything is working perfectly when i'm authenticating a sudo command in the terminal. 
_[The set up in fedora is a little trickier than in ubuntu, but after some dependencies fixing / pam.d modify everthing works]_

The problem is in the login screen : Howdy starts, but it shows _"unknown error: 1"._

Doing some debugging in the code, i realized that:

Howdy is starting correctly.
It calls  `/lib64/security/howdy/pam.py` then at line 43 (when the status variable is defined from the call at compare.py) something strange happens: 
The returned value by `subprocess.call(...)` is 1.
Reading the code the only case in which  compare.py  returns 1 is if it can't find the data file in dlib. 
So i changed the return value  in compare.py where the check is done AND in the log screen it still prints  "unknown error: 1" (where 1 is the value of status, because it's not changed anywhere else in the code). 

From this i deduced that the problem is in the subprocess call that can't be executed correctly in the log screen, but only once the log has been done; (after suspend same error)

[Has i often found in the issue changing the permission of the howdy folder is not fixing (chmod -R  o+x  etc..)]

I know that the same "issue" has been reported already, but i have not seen an analysis of the problem yet (nor a solution).

Linux distribution: Fedora 34 

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2021-05-13:
You've done some good digging! Error 1 is a general failure, can be triggered anywhere. `/var/log/auth.log` could have more details about what exactly went wrong. 90% of the time this is a permissions issue.

gabrieleaquaro, 2021-05-13:
I forgot to say that i'm missing the log file that should be at `/var/log/auth.log` (or at least i could not find it anywhere, don't know why, it's a fresh install though).

Anyway i'm pretty sure that the point "crash" is the subprocess call , because any change in the code of `compare.py`

gabrieleaquaro, 2021-05-13:
I forgot to say that there is no log file (i don't know why, it's a fresh install and i looked pretty kuch everywhere).

Anyway i'm pretty sure that the point of failure is the call at the subprocess. Because any change in compare.py has no effect on that error. 

If you have any idea of how to find the log file, because `/var/log/auth.log` doesn't exist. (At least in Fedora)

Sorry for the close, but i was replying from mobile and the buttons are close.

gabrieleaquaro, 2021-05-16:
Ok i managed to make it work. Apparently was a prblem of modules in SELinux in Fedora.
I leave here the solution [HERE!:](https://linuxreviews.org/Howdy/SELinux). 
Thanks for Help by the way, great job with this repo !

closed 2021-05-16 by gabrieleaquaro

boltgolt, 2021-05-16:
Thanks for figuring it out and for linking that site, i've never read the [page on howdy](https://linuxreviews.org/Howdy) but the snarky comments about "stupid problems" (without opening a pull request for them) is hilarious