boltgolt/howdy #545: IRC/Discord channel

AwesomestCode, 2021-05-15:
Would appreciate if there was a community style real-time support channel. I think it would help users like myself with basic questions about how Howdy works or assistance with installation (I wanted to know how this fares against Windows Hello and Apple Face ID in terms of security).

boltgolt, 2021-05-15:
It's a good suggestion but it has one big caviat: On GitHub all issues are easily searchable and even then about half the opened issues boil down to the same 5 problems. I'm afraid having a chat channel would basically just be people asking how to set the device path

AwesomestCode, 2021-05-15:
Have an FAQ bot that can be used to provide pre-written answers. That's what we do at PrismarineJS.

boltgolt, 2021-05-16:
I guess it would work a lot better projects that have more of a community, like prismarine. I would think there's not a whole lot of people that want to be in a community of their authentication provider. I might be wrong of course so if people do want that, leave a comment here

EmixamPP, 2021-06-03:
GitHub has recently implemented a "discussion" feature. It can be activate in the "Settings" tab. 
[image]

boltgolt, 2021-06-06:
I've seen it used but i'm not 100% sure what it adds over just good old issues, as far as i'm aware it's just issues with a different UI