boltgolt/howdy #549: I cant add a new face model.

ThisUsernameWasNotTakenButNowItIs, 2021-05-29:
I cant add a new face model. When I run the command: "sudo howdy add", it gives me this error: 
"Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7efc857df550>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release(internal)
AttributeError: 'VideoCapture' object has no attribute 'internal'". 

I have not searched for similar issues, but I'm pretty sure that the issue has not been reported yet.

Linux distribution: Ubuntu 20.04

Howdy version: 2.6.1

Please tell me how to fix it. Thanks.

ThisUsernameWasNotTakenButNowItIs, 2021-05-29:
And I'm sorry, but I may not be able to respond.

boltgolt, 2021-05-29:
Please search for exsiting issues first. From your output: `Camera path is not configured correctly, please edit the 'device_path' config value.`

closed 2021-05-29 by boltgolt