boltgolt/howdy #552: Is Howdy compatible with the XPS 13 9310?

logicito, 2021-06-02:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I will install Howdy and the IR emiters will turn on fine and face recognition will work, nonetheless when I reboot or shutdown the laptop, when trying to login again, the light from the camera will turn on (white) but the IR emiters will be off, and they will never work again, any idea what could be the problem?

Thank you

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

EmixamPP, 2021-06-03:
I don't think it's related to Howdy. But rather a problem with the kernel.
But don't worry I've the solution to your problem ! 

Take a look at my repo : https://github.com/EmixamPP/linux-enable-ir-emitter and read the `Automatic setup` part. 
Normally you shouldn't have any problems, but if you need me I'll be available.

Doomsdayrs, 2021-06-20:
This issue can be closed.

closed 2021-06-20 by boltgolt