boltgolt/howdy #554: Login keyring did not get unlocked when I (Howdy), logged in [Ubuntu 20.04.2 LTS]

HorseboxHeaven, 2021-06-07:
When I run a sudo command, Howdy recognizes me and provides the password. I have created 6 face models and Howdy seems to work pretty well in different light levels and face angles, except for one thing. Howdy allows the PC to complete the boot and load the desktop. The startup applications load, and then I get an authentication prompt advising me that the login keyring did not get unlocked when I (Howdy), logged in.  So after all, I still need to use my password to access the PC.  I have a twin screen setup and here's what I see.

closed 2021-06-08 by boltgolt

boltgolt, 2021-06-08:
Doesn't happen on every installation, only when using programs that actually utilize the keyring like chrome

HorseboxHeaven, 2021-06-08:
I'm not using Chrome. Or do you mean when Chrome is simply installed on the machine?

boltgolt, 2021-06-08:
It just has to be installed and there are quite a few other programs that use it. It's a bit besides the point though, if I listed all possible complications on the readme it would be 4 times as long