boltgolt/howdy #556: Failed to read camera specified in your 'device_path', aborting

hilfing, 2021-06-26:
PLs help.
The output of v4l2-ctl --list-devices is: -
`UVC Camera (046d:0825) (usb-0000:00:14.0-7):
	/dev/video0
	/dev/video1
`
I tried putting both in the device path by doing `sudo howdy config`. but none works. pls help

boltgolt, 2021-06-29:
What's the full output howdy generates?

hilfing, 2021-06-30:
```root@hilfing-PC:/home/hilfing# howdy add
Adding face model for the user hilfing
Enter a label for this new model [Initial model] (max 24 characters): glasses1

Please look straight into the camera
Failed to read camera specified in your 'device_path', aborting
```

hilfing, 2021-07-07:
Pls help

u473t8, 2021-09-19:
Seems to be similar issue on login attempt, Fedora 34
Here is `journalctl output`

```
Sep 19 02:52:15 fedora [11865]: Attempting facial authentication for user u473t8
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/videoio_registry.cpp (217) VideoBackendRegistry VIDEOIO: Enabled backends(9, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); MSMF(970); V4L2(960); CV_IMAGES(950); CV_MJPEG(940); FIREWIRE(930); UEYE(920)
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (324) getPluginCandidates VideoIO plugin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (331) getPluginCandidates     - /lib64: 0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (335) getPluginCandidates Found 0 plugin(s) for FFMPEG
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (831) open OpenCV | GStreamer: /dev/video0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (864) open OpenCV | GStreamer: mode - FILE
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ WARN:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ WARN:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ WARN:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (324) getPluginCandidates VideoIO plugin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (331) getPluginCandidates     - /lib64: 0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (335) getPluginCandidates Found 0 plugin(s) for INTEL_MFX
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (324) getPluginCandidates VideoIO plugin (MSMF): glob is 'libopencv_videoio_msmf*.so', 1 location(s)
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (331) getPluginCandidates     - /lib64: 0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/backend_plugin.cpp (335) getPluginCandidates Found 0 plugin(s) for MSMF
Sep 19 02:52:16 fedora audit[11869]: AVC avc:  denied  { map } for  pid=11869 comm="python3" path="/dev/video0" dev="devtmpfs" ino=503 scontext=system_u:system_r:xdm_t:s0-s0:c0.c1023 tcontext=system_u:object_r:v4l_device_t:s0 tclass=chr_file permissive=0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ WARN:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_v4l.cpp (833) createBuffers VIDEOIO(V4L2:/dev/video0): failed mmap(614400): errno=13 (Permission denied)
Sep 19 02:52:16 fedora gdm-session-worker[11869]: Failed to read camera specified in your 'device_path', aborting
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_images.cpp (282) icvExtractPattern Pattern: /dev/video%01d @ 0
Sep 19 02:52:16 fedora gdm-session-worker[11869]: [ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/cap_images.cpp (313) open CAP_IMAGES: Stop scanning. Can't read image file: /dev/video0
Sep 19 02:52:16 fedora [11865]: Failure, unknown error1
```

u473t8, 2021-09-19:
Answer in the bottom of https://copr.fedorainfracloud.org/coprs/principis/howdy/ solved problem.

closed 2022-04-28 by boltgolt

kylebakerio, 2022-11-14:
I'm getting this issue when attempting to install following [instructions from system76](https://support.system76.com/articles/setup-face-recognition/) on my x1 extreme gen 4.

I've tried selecting all my camera that show up (/dev/video0,12,3).

I never see the camera light come on when running howdy test with any of those values.

I get the same error message mentioned at top here when setting device to one of those cameras.
Also note: before I did that, I got a different error message that told me to edit the device_path config value.

kylebakerio, 2022-11-14:
Cheese app shows the four cameras as selectable, and shows 0 and 2 as B/W and 1 and 3 as Color. (I don't know why it lists as four cameras, as far as I can tell I have only one built in to the top of the lid, presumably there is a second IR one in there as well; are there really four cameras there to get a 3d face model or something...?)

kylebakerio, 2022-11-14:
> Answer in the bottom of https://copr.fedorainfracloud.org/coprs/principis/howdy/ solved the problem.

I tried checking here but wasn't able to see anything worth trying that I was able to understand...

kylebakerio, 2022-11-14:
aaaand like an idiot, I just tried with 'sudo' and it worked. The instructions I was following didn't mention sudo. Guess a more helpful error message for lack of permissions would be helpful. Thanks for this, I'll leave these here for the next idiot to run into this.