boltgolt/howdy #557: howdy working on terminal sudo, but not working with pkexec (gui-based sudo) authentication on ubuntu 21.04

thiagomoreirac, 2021-06-27:
I noticed this behavior while trying to update the system through the gui ubuntu updater. 

Howdy works properly on the gdm and terminal sudo. However, while trying to use gui-based sudo authentication, like synaptics or updater, the pkexec popup will show "face detection authentication" and will loop between  a failed / face detection msg in the pkexec. I see the camera led blinking quickly. Therefore, pkexec doesn't fall back to fingerprint/password authentication, which forces me to disable howdy from pam auth. 

----

I've searched for similar issues already, and my issue has not been reported yet. (checked)

Linux distribution (if applicable): ubuntu 21.04

Howdy version (`sudo howdy version`): 2.6.1

kolbma, 2021-07-03:
See #561

vega-d, 2022-04-01:
I am experiencing the same problem on 2.6.1, but on Fedora Silverblue. 
Are there any troubleshooting steps, or log files I can look into?

nuevoparadigma, 2022-11-10:
To use a password or a face in a graphical interface,
add the following lines to the beginning of any required file, inside /etc/pam.d/* (*su, *su-l,*sudo, *sudo-i):

   auth sufficient pam_unix.so try_first_pass likeauth nullok
   auth sufficient pam_python.so /lib/security/howdy/pam.py

This will first ask for a password.
Then pressing Enter on a blank field will proceed to face authentication.

closed 2026-11-16 by boltgolt