boltgolt/howdy #560: Illegal instruction is not fixed

kolbma, 2021-07-03:
When running `sudo howdy add` or `sudo howdy test` the only output is `Illegal instruction`.

----

I've searched for similar issues already, and my issue has been reported, but closed without solution in current release:
#101 
#330 

Linux distribution (if applicable): openSUSE Leap 15.2

Howdy version (`sudo howdy version`): Howdy 2.6.1

kolbma, 2021-07-03:
I can't use the provided python3-dlib rpm. I've installed now the current release with pip3 install dlib.  
First with `pip3 install --user dlib` to get a wheel.  
Afterwards `sudo pip3 install .cache/pip/wheels/bd/46/7c/d...ce/dlib-19.22.0-cp36-cp36m-linux_x86_64.whl` and `pip3 uninstall dlib`.

Seems to work now. Maybe update your documentation and try to catch this error with a meaningful error message!

boltgolt, 2021-08-13:
I'm guessing that it crashes Howdy on cv2 import, might not be a catchable error but i'll look into it

jcdegiorgi, 2022-08-08:
Hi,
I've got this same problem when sudo howdy -U myusername add
_Please look straight into the camera
Illegal instruction_

But the howdy test works perfectly though.

Ubuntu 22.04 and HP Pavilion dv3000.

Regards,
Jean-Claude

kolbma, 2022-10-27:
I've found some time to look into this...  
`dlib` is using some CPU architecture instructionset features like __SSE4__ or __AVX__.  

In precompiled binaries (on modern hardware) this __defaults__ to enabled __AVX__ instructions.  
When running this `dlib` on older hardware these are illegal instructions for the CPU.  

But when installing/compiling `dlib` on the local hardware with `pip`, `cmake` is detecting the correct available instructionset and uses right features for compiling `dlib`. So e.g. using __SSE4__ instead of __AVX__ or disabling both and using slower generic machine instructions.

It is also possible to set this by hand: 
https://github.com/davisking/dlib/blob/1bebccfb8728f6112d6142ba24f2dd65970ff6a2/setup.py#L20-L22
```
To exclude certain options in the cmake config use --no:
    for example:
    --no USE_AVX_INSTRUCTIONS: will set -DUSE_AVX_INSTRUCTIONS=no
```

In `dlib` is also a check for the instruction sets...  
https://github.com/davisking/dlib/blob/1bebccfb8728f6112d6142ba24f2dd65970ff6a2/dlib/simd/simd_check.h#L153-L171
```C
    inline void warn_about_unavailable_but_used_cpu_instructions()
    {
#if defined(DLIB_HAVE_AVX2)
        if (!cpu_has_avx2_instructions())
            std::cerr << "Dlib was compiled to use AVX2 instructions, but these aren't available on your machine." << std::endl;
#elif defined(DLIB_HAVE_AVX)
        if (!cpu_has_avx_instructions())
            std::cerr << "Dlib was compiled to use AVX instructions, but these aren't available on your machine." << std::endl;
#elif defined(DLIB_HAVE_SSE41)
        if (!cpu_has_sse41_instructions())
            std::cerr << "Dlib was compiled to use SSE41 instructions, but these aren't available on your machine." << std::endl;
#elif defined(DLIB_HAVE_SSE3)
        if (!cpu_has_sse3_instructions())
            std::cerr << "Dlib was compiled to use SSE3 instructions, but these aren't available on your machine." << std::endl;
#elif defined(DLIB_HAVE_SSE2)
        if (!cpu_has_sse2_instructions())
            std::cerr << "Dlib was compiled to use SSE2 instructions, but these aren't available on your machine." << std::endl;
#endif
    }
```

This gets called by the python module here:
https://github.com/davisking/dlib/blob/1bebccfb8728f6112d6142ba24f2dd65970ff6a2/tools/python/src/dlib.cpp#L42

I've opened https://github.com/davisking/dlib/issues/2675

boltgolt, 2022-10-31:
Awesome work @kolbma! Hopefully the dlib people can look at it soon

closed 2022-10-31 by boltgolt