boltgolt/howdy #563: Howdy isn't a command

Fancy2209, 2021-07-06:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._
Hi, after a bit of problems I finally got howdy to install without failing, but, now the thing is that I can't type sudo howdy or howdy, it just says it isn't a command, how do I fix this?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Pop!_OS 21.04

Howdy version (`sudo howdy version`): Can't use the howdy command

Fancy2209, 2021-07-06:
What's even weirder is that `man howdy` works

ShabazzKing, 2021-08-03:
> _Please describe the issue in as much detail as possible, including any errors and traces._
> _If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy [...]

I had the same problem (and many other problems, which made me uninstall howdy by the way). Perhaps this helps you. What I did is:
cd /usr/local/bin
sudo ln -s /usr/lib/security/howdy/cli.py howdy

closed 2026-11-16 by boltgolt