boltgolt/howdy #564: FileNotFoundError: [Errno 2] No such file or directory: 'pip3'

yama-yeah, 2021-07-07:
I tried to install howdy on ubuntu20.14.
but, i can't it.
console show this.
----
$sudo apt install howdy
--------omission---------
>>> Upgrading pip to the latest version

Requirement already satisfied: pip in /usr/lib/python3/dist-packages (20.3.4)
Collecting pip
  Using cached pip-21.1.3-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
  WARNING: The scripts pip, pip3 and pip3.9 are installed in '/usr/local/bin' which is not on 
PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-
warn-script-location.
Successfully installed pip-21.1.3
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behav
iour with the system package manager. It is recommended to use a virtual environment instead: 
https://pip.pypa.io/warnings/venv

>>> Upgrading numpy to the latest version

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 116, in <module>
    handleStatus(subprocess.call(["pip3", "install", "--upgrade", "numpy"]))
  File "/usr/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3'

dpkg: An error occurred while processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing a trigger for doc-base (0.11.1) ...
Processing one more doc-base file...
Processing libc-bin (2.33-0ubuntu5) trigger ...
Processing man-db (2.9.4-2) trigger ...
An error occurred during processing:
 howdy

E: Sub-process /usr/bin/dpkg returned an error code (1)

----
I am Japanese, so some of the messages displayed in the console are in Japanese.
So I translated the Japanese messages back into English once. It may be that some of the messages have changed your message.
sorry.
why can't i install howdy?
Maybe it's because I'm using pyenv?
please help me.
Sorry for my poor English.
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): ubuntu 20.14

Howdy version (`sudo howdy version`): 2.6.1

yama-yeah, 2021-07-07:
I installed it from gdebi without using apt and it worked.
I do not know what is causing this.

boltgolt, 2021-08-14:
Your issue are these lines:

```
Successfully uninstalled pip-20.3.4
WARNING: The scripts pip, pip3 and pip3.9 are installed in '/usr/local/bin' which is not on
PATH.
Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-
warn-script-location.
```

This is telling you that pip is installed, but you can't use it because it's not in the PATH. Your setup where `/usr/local/bin` is not in PATH is highly unusual, you might want to fix that. Good to hear you got it working anyway

closed 2021-08-14 by boltgolt

yama-yeah, 2021-08-14:
Thanks for the reply.
Now that we have facial recognition, logging in is a hell of a lot easier!
Thank you!

danielk011, 2021-09-06:
I don't think the issue is with /usr/local/bin not in the PATH. I am also having the same issue (in Ubuntu 21.04), but my path has /usr/local/bin: "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin". I'm glad the installation still works with gdebi though.