boltgolt/howdy #565: Behavior of config variable ignore_closed_lid is inverted

redcubie, 2021-07-07:
Current behavior is when ignore_closed_lid is true, the lid state is checked and when it's false, the lid state is ignored. The expected behavior is when an "ignore" variable is true, the thing gets ignored.

FIX:
change line 30 of pam.py to `if not config.getboolean("core", "ignore_closed_lid"):`
OR
change all occurrences of ignore_closed_lid to sense_closed_lid

boltgolt, 2021-07-07:
Yeah this was discussed at the time of implementing, and is caused by the prior naming of `ignore_ssh`. Both of them were named from the perspective of PAM, so whether PAM should ignore (skip) the module if the lid is closed. I agree it's not a very intuitive way to name them, i'm open for suggestions. Personally, i don't think "sense_closed_lid" would be ideal either, as it doesn't just sense the state, it stops execution

redcubie, 2021-07-07:
Another option would be to rename it to `ignore_when_lid_closed` or `cancel_when_lid_closed`

boltgolt, 2026-11-16:
Hey, thanks for reporting this! In v3 this option was renamed to `abort_if_lid_closed` (and `ignore_ssh` to `abort_if_ssh`), so the name now matches what it does. Closing this as fixed!

closed 2026-11-16 by boltgolt