boltgolt/howdy #567: Reference my repo in the wiki

EmixamPP, 2021-07-10:
I have very recently produced a utility that allows you to configure any of your infrared cameras (unless they are not supported by the Kernel). 

And regularly people ask "why doesn't it work" when all they need to do is run a single command with my program : https://github.com/EmixamPP/linux-enable-ir-emitter. 

Sometimes, I  take the time to look at the issues and redirect them to my repo. So I thought if the wiki page for this project mentions my repo it might be useful for a lot of new Howdy users.

EmixamPP, 2021-08-31:
I took the liberty of doing it myself, if you find it inappropriate, don't hesitate to delete it from the wiki, I apologize in advance !

boltgolt, 2021-08-31:
Sorry, i thought i responded already! Brilliant project and the spot on the common issues page is more than deserved!

I've been looking through your project and i think it might be good to have as an optional dependency for Howdy, i think it will smooth installation for quite a few people. The next version of Howdy has a GUI that guides a user through configuration and also checks the camera for them. Maybe your solution could be called from the GUI somewhere during installation?

For 3.1.0 i could package `linux-enable-ir-emitter` into the Howdy debian PPA so users can install it as an recommended dependency together with Howdy

EmixamPP, 2021-09-02:
I can also implement an `-y` option to my command in order to skip all questions. The chances of a user inputting 'no' are low. In this case, we could do without the terminal and only rely on probabilities.
If we do without a terminal, I could also make more error codes in order to better inform the user of possible problems encountered.

EmixamPP, 2021-09-04:
I think it’s better to run linux-enable-ir-emitter only for users who have problems. And for me the most important reason is that some user don't want a useless installed software.
More over, by bad luck it may break (until the next standby) the ir blaster. But again, the chances are very very low. 

And yes we can do it like that, the emitter is triggered, if it does not work, linux-enable-ir-emitter is called. I set up a complete exit code system so I don’t have to display a terminal. If the success code is returned, Howdy can ask again if it worked.

The version 3.0.0 of Howdy is on the dev branch ?

boltgolt, 2021-09-04:
That work work. The UI already elevates privileges graphically so I think it won't be an issue to run your program underneeth. But your package would be installed already regardless of emitters if we make it a recommended package. We'd have to run an apt command to install it dynamically I guess 

3.0.0 is currently on the beta branch, there's a large file structure difference

EmixamPP, 2021-09-16:
I have a lot to say so I'll try to be concise. 

Version 3.0.0 of `linux-enable-ir-emitter` should be released soon. I rethought all its functioning and I arrived at one of my most beautiful goals: no more need for Windows, no more need to follow a tuto, no more need for the community to share its configurations, no false positive, no danger for a working camera, etc ! Any infrared camera (unless it is not recognized by the kernel) can really be configured automatically.

However, it is now mandatory that the user answers yes/no to the questions the terminal asks. So, do you think it's possible that if the user enters in your gui that his blaster is not working: an external terminal should be opened with the command `linux-enable-ir-emitter configure`?  And eventually in case of an error code, the user is sent back to my GitHub.

I could eventually develop a system a bit like yours, to analyze the brightness of the frames to determine if the ir emitter is working or not. But that's a lot to avoid a simple yes/no input and the brightness of the room could also bias the result. Besides the fact that it is out of my skills. 

That's it, so I think that this new version of my software will suit your project very well.

EmixamPP, 2021-09-28:
And the 3.2.0 of `linux-enable-emitter` will support multiple infrared camera configuration. (I don't know if it is possible with Howdy, but in case it will be fine)

Version 3.3.0 will make it possible to configure the ir emitter without questions in the terminal. However I could not give a release date.

EmixamPP, 2023-06-23:
#683

closed 2023-06-23 by EmixamPP