boltgolt/howdy #568: Wrong directories permissions in Debian package

tikhont, 2021-07-11:
Following directories in /lib/security/howdy don't have execution permission bit after installation: dlib-data, pam-config, recorders. So screensaver unlock doesn't work.

This command helped:
sudo chmod o+x /lib/security/howdy/{dlib-data,pam-config,recorders}

/var/log/auth.log messages:
Jul 11 08:45:06 *** [HOWDY]: Attempting facial authentication for user ***
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]: Traceback (most recent call last):
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]:     return doAuth(pamh)
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]:   File "/lib/security/howdy/pam.py", line 85, in doAuth
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Jul 11 08:45:06 *** /lib/security/howdy/pam.py[20536]: PamException: Conversation error

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Debian, LMDE

Howdy version (`sudo howdy version`): 2.6.1

Mckol, 2021-08-11:
Thank you! This fixed howdy in `gnome-screensaver` for me! 
(I think I also needed to disable the "Attempting Face Recognition" message in howdy config)

boltgolt, 2021-08-11:
Unfortunately this is not a bug with Howdy but with the PAM implementation of the screensaver. You can use the known workaround you posted to do a temporary fix.

closed 2021-08-11 by boltgolt