boltgolt/howdy #574: Can't open device Tobii Eye Tracker 4C

Benibla124, 2021-07-23:
Linux distribution: Arch Linux 5.13.4-arch1-1

Howdy version: Howdy 2.6.1

Hello there, I'm trying to set up Howdy with my Tobii Eye Tracker 4C, but Howdy can't seem to open the device. I have tried setting the device_path variable to both `/dev/video0` and `/dev/video4` (those show up when running `v4l2-ctl --list-devices` but in both cases `sudo howdy test` doesn't work.

If I try with my Logitech C920 everything works fine.

boltgolt, 2021-08-09:
Can you access those devices from other apps such as cheese?

Benibla124, 2021-08-10:
No, I can't.

boltgolt, 2021-08-10:
Then your camera is probably not supported unfortunately. You could try updating your kernel but seeing how specialized this camera is it might just not use the encodings Howdy needs.

closed 2021-08-10 by boltgolt