boltgolt/howdy #577: Migrate Model from dlib models to Deepface

ashvinnihalani, 2021-08-03:
Could we migrate the model from the dlib models to Deep face? It is more up-to-date and could potentially improve the accuracy of the detection? There could be an option to switch from one to another if there are resources or other contraints.

I could help with the migration and testing if you need it.

boltgolt, 2021-08-05:
Sounds interesting, especially because v3 will introduce breaking changes anyway. In your personal opinion, what improvements would switch models cause?

ashvinnihalani, 2021-08-05:
Out of the box it allows for just a wide variety of models including their
own implementation of a Dlib model. In addition it seems like accuracy is
much higher. Davisking/dllib models seem to have an accuracy of 99.3 and
the highest model in Deepface, FaceNet /w 512d got a 99.65. The delta in
the performance is small enough that we should probably test this with a
similar data-set since I am unsure which dataset these metrics are deriving
from. In addition, there is a detect, align, represent and verify pipeline
that they have implemented under the hood that might be useful and might
reduce the number of false negatives we get from differing positions of
heads.
<https://scontent-sjc2-1.xx.fbcdn.net/v/t1.0-9/63870_148996021805559_6945915_n.jpg?oh=3803dae3f8804f293f70330c1cbb281a&oe=5931A264>
Ashvin Nihalani
UC Berkeley - 2020
832-628-6104

<https://facebook.com/ash.nihalani.9> <https://github.com/ashvinnihalani>
<https://www.linkedin.com/in/ashvinnihalani/>

boltgolt, 2021-08-09:
My worry is that we will be rewriting the absolute core of Howdy without much benefit. Our dlib implementation does already detect (`face_detector`), align (`pose_predictor`), represent (`compute_face_descriptor`?) and verify (`numpy`) so i'm not sure if that's an improvement. I guess Deepface has a better success rate there?

I think i would like to see a demonstration of how much better it is first, just to be sure we're not doing unnecessary work and introducing bugs into heavily field-tested code.

boltgolt, 2021-08-09:
(also, the ties to Facebook are less than ideal for a project dealing with face recognition unfortunately)