boltgolt/howdy #579: Ubuntu 21.04  File "/lib/security/howdy/compare.py", line 17, in <module>

t3rmin4l, 2021-08-17:
Just upgraded to 21.04 and getting this error. Did `pip install dlib`, did https://www.pyimagesearch.com/2018/01/22/install-dlib-easy-complete-guide/ - still same.

JoeFromDenver, 2021-09-22:
Pop OS 21.04 here. I had the same issue with that dlib error. I removed Howdy completely and installed from scratch. Went perfectly and now I'm back in.

Svetlana-T, 2022-03-19:
```
sudo pip3 install dlib
```

says this

```
Collecting dlib
  Downloading dlib-19.23.1.tar.gz (7.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.4/7.4 MB 3.2 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 14, in <module>
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 20, in <module>
          from setuptools.dist import Distribution, Feature
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 35, in <module>
          from setuptools.depends import Require
        File "/usr/lib/python3/dist-packages/setuptools/depends.py", line 7, in <module>
          from .py33compat import Bytecode
        File "/usr/lib/python3/dist-packages/setuptools/py33compat.py", line 55, in <module>
          unescape = getattr(html, 'unescape', html_parser.HTMLParser().unescape)
      AttributeError: 'HTMLParser' object has no attribute 'unescape'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

Debian

boltgolt, 2022-03-19:
@Svetlana-T completely different issue and terminal output. Please use google: https://stackoverflow.com/questions/65640476/pip3-on-python3-9-fails-on-htmlparser-object-has-no-attribute-unescape

closed 2026-11-16 by boltgolt