boltgolt/howdy #580: Unable to install howdy under Ubuntu 20.04

hmerovich, 2021-08-20:
Hi

I have trying to install howdy under Ubuntu 20.4 but each time I run

sudo apt install howdy 

my machine power off 

How can I diagnose the problem?

Best regards and thanks for your help.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

hmerovich, 2021-08-21:
I tried again to install howdy and now I could install it but when I tried
sudo howdy add
I get an import error

I am attaching the screenshot
[image]

Thanks you very much and best regards

ntfrueh, 2021-08-23:
I was also having issues after installing Howdy yesterday on ZorinOS (Ubuntu 20.04) - got the  "command not found".

Tried purging the app, cleaning/autoremoving apt, then finally re-installed successfully (where I saw it downloading Numpy packages), so I guess it was just a blip in my initial install process.

I'd say try the following, then see if you get a different error (cause it seems like some missing/improperly installed dependencies somewhere)

`sudo apt purge howdy`
`sudo apt autoremove`
`sudo apt update`
`sudo apt install howdy`

boltgolt, 2021-08-23:
A spontaneous shutdown is _new behavior_ for sure. The installer runs a c compile that can take quite a lot of power, maybe your power supply is not up to the task?

The second issue is a corrupted install, the steps from @ntfrueh will help there

closed 2026-11-16 by boltgolt