boltgolt/howdy #586: Conflict issue with Python & Fortran.

notbluu, 2021-09-05:
Tried installing through the PPA, but I get this error:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgfortran5 : Depends: gcc-11-base (= 11-20210417-1ubuntu1) but 11.1.0-1ubuntu1~21.04 is to be installed
 libpython3.9-dev : Depends: libpython3.9-stdlib (= 3.9.4-1) but 3.9.5-3~21.04 is to be installed
                    Depends: libpython3.9 (= 3.9.4-1) but 3.9.5-3~21.04 is to be installed
 python3.9-dev : Depends: python3.9 (= 3.9.4-1) but 3.9.5-3~21.04 is to be installed
                 Depends: libpython3.9 (= 3.9.4-1) but 3.9.5-3~21.04 is to be installed
E: Unable to correct problems, you have held broken packages.

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Ubuntu 21.04

Howdy version (`sudo howdy version`):

boltgolt, 2021-09-05:
All of those packages suggested by the debian installer would work with Howdy and the specific versions apt seems to have a problem with are not a requirement. Howdy simply needs `python3` and `python3-dev` but the version is irrelevant. 

Are you sure you do not have a broken package registry? Maybe running `sudo apt --fix-broken install` will help?

notbluu, 2021-09-07:
It's a broken package somewhere, I switched over to popOS and it works fine now.

closed 2021-09-07 by notbluu