boltgolt/howdy #587: Change profile from fast to Secure

shahulhameed8, 2021-09-07:
Howdy can work in three mode: Fast, Balanced, and Secure. While installing the app, you’ll be able to choose between them.  

Fast – speeds up the scanning process greatly, but it allows more fuzzy matches. 
Balanced – relatively quick detection, but might not log you in when further away. 
Secure – is the safest option, thought it will take much longer to recognize your face.

Can anyone please tell me how can i change fast profile to secure profile after installation of howdy in pop os.

boltgolt, 2021-09-07:
Just change the `certainty` config option to a lower value, like 3.5

closed 2021-09-07 by boltgolt

shahulhameed8, 2021-09-08:
> Just change the `certainty` config option to a lower value, like 3.5

Thank you for the response bro. One more question bro. 
So certainty Fast is 4.2
Balance is in between 3.5 to 4.2
Secure is 3.5 like that?

boltgolt, 2021-09-12:
You can see the values used in the installer [here](https://github.com/boltgolt/howdy/blob/master/debian/preinst#L140)