boltgolt/howdy #591: Script while recognizing faces

KrishnanS2006, 2021-09-20:
As described [here](https://github.com/boltgolt/howdy/issues/220), I am attempting to use Howdy while my webcam is turned sideways. I have a working script that, when run, will stream the correctly oriented camera  to a new capture device (`/dev/video4`). However, ffmpeg has a hard time running in the background, and I am also worried about CPU usage with a script like this running in the background. Is there any way (config file, editing another file, etc.) to run a script during the face recognition period, and then stopping it at the end? Or at least, running it as soon as Howdy starts and stopping it after it finishes running? Having this feature would be amazing and would give users with Linux experience much more customization. And if this feature will probably not be implemented, is these some way I can "hack" this terminal command somewhere into Howdy? Thanks for the help!

Rolodo, 2022-06-21:
I would also like this feature using my surface I need to manualy start the cam with a script, and I would not like to always be on.

Thanks

boltgolt, 2022-06-21:
I guess a good step would be to understand the way PAM config files work. [This wiki page](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types) might push you in the right direction.

Once you understand that you can go ahead and manually add howdy to a file, probably `common-auth` for your purposes. You can then add a script just before Howdy like this:

```
auth       optional     pam_exec.so seteuid /path_to_your_script.sh
auth       sufficient   pam_python.so  /lib/security/howdy/pam.py
```

KrishnanS2006, 2022-06-24:
@Rolodo could you please share? I would love to know how you went about doing this...