boltgolt/howdy #606: Cannot reinstall howdy

Prophet523, 2021-10-31:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Attempted to install Howdy but it wouldn't pull the keyring when logging in. Tried to uninstall and reinstall to go through a step I skipped in the setup but everytime I try to reinstall I get this error

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 110, in <module>
    handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
  File "/usr/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3'
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

All attempts to finish the installation have failed. I have tried doing  sudo dpkg --configure -a 

it will look like its continuing but howdy never actually installs.

What am I doing wrong here?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Pop-OS 21.04

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2021-10-31:
The keyring is a wontfix issue unfortunately. Your keyring is encrypted
with your password, so by using Howdy (and not entering your password) it
can't be unlocked. There's currently sometime very wrong with pip3 that's
breaking installations, take a look at the other issues

stavro-venustech, 2022-03-21:
there must be a way to remove the file right? (apart from reinstalling the os)

stavro-venustech, 2022-03-21:
It seems like it will work when you think about it

stavro-venustech, 2022-03-21:
where is the directory where the keyring is stored, I might be able to remove/edit the keyring which should therefor fix the problem, unless that makes it worse ;)

Cody-Q, 2022-04-14:
https://github.com/boltgolt/howdy/issues/621#issuecomment-985109099
Had the same issue and ran these two commands to fix it