boltgolt/howdy #607: Howdy cannot recognize my face

bbbear2002, 2021-11-03:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Howdy cannot recognize my face.

Howdy installed successful. Add face successful. But it cannot recognize my face. When i run a howdy test, it always shows a red circle. How to solve this issue? Thanks.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

lnee94, 2021-11-05:
can you give some screen shots?

Bram-diederik, 2021-11-26:
You can look to the snapshots pictures with probability at `/usr/lib/security/howdy/snapshots/`

bbbear2002, 2021-12-01:
Thank you both. My screenshot is here: 

[image]

bbbear2002, 2021-12-01:
[image]<br /><a target='_blank' href='https://statewideinventory.org/volkswagen-0-60-times'>2013 volkswagen cc 0 60</a><br />

It shows successful login but login failed.

Bram-diederik, 2021-12-01:
I had problems with the default howdy settings. Disabled those and put it into pam.d directly.

'auth    sufficient pam_python.so /lib/security/howdy/pam.py'

EmixamPP, 2021-12-04:
It seems that you don't use an infrared camera.  (Your camera isn't in gray scale)
Use an infrared camera is particularly recommended for facial authentication.

closed 2026-11-16 by boltgolt