boltgolt/howdy #608: [Asus ZenBook Q404IQ] Use both infrared and normal camera

androidacybot, 2021-11-08:
Hello!

On windows, Windows Hello uses both cameras (both the IR emitters and the camera light come on at once) for face unlock, and it's snappy and accurate. However, this utility only seems to support using one at a time, which leads to a very bad experience.

Just using the IR emitters will register fine, but won't recognize almost ever, no matter how much messing with config I do. Just the camera works ok, but is less accurate and easier to fool.

Is there any way to get it to work like it does on windows, ie using both cameras in tandem?

boltgolt, 2021-11-10:
Does Windows Hello use both cameras? I've only ever read about the IR camera use on Microsoft articles. Either way, the current limit on Howdy recognition speed is your CPU, so adding another camera would slow down things even more.

Either way, an easier fix is playing with your certainty config value, your matches are probably juuust not certain enough.

EmixamPP, 2021-11-12:
You think that Windows uses the both camera because when you train the recognition, the video feedback is in colour ? 
But this not means that the "regular" camera is used to recognize your face.

androidacybot, 2021-11-12:
No, if you'll reread what I said, I think that because _both hardware light indicators light up_. The camera light AND the infared emitter lights come on, continuously. With howdy I'm limited to one hardware device, and Linux recognizes the IR emitter and camera separately.

EmixamPP, 2021-11-12:
When you said `continously` you mean that the infrared emitter blink (in red) and the camera utilization indicator lights (in white). 

When you use Howdy, which of these two works ? (You didn't specify this in your comment) 
And which camera (descriptor/path) is used in Howdy ?

boltgolt, 2021-11-12:
To add to that, both the normal white "camera in use" light and the IR emitters activate when only using the IR camera on either of my Windows Hello laptops

androidacybot, 2021-11-13:
Both "work" but infrared on its own has to be lowered to an insanely low accuracy level to recognition after registering and the main camera has a lot of false positives.

There's /dev/video[1-4] but only 1 and 3 work.

And if the IR emitter was the only one in use - why doesn't just IR have the same accuracy as windows? Windows is both fast and accurate, with every indicator I can find indicating that both are in use, and on Linux I get either fast at best and almost never accurate.

I'm completely willing to blame Asus fuckery for this - heaven knows I've hit that with their dual GPU setup lol.

EmixamPP, 2021-11-13:
> And if the IR emitter was the only one in use - why doesn't just IR have the same accuracy as windows? Windows is both fast and accurate

It depends on the neural network used (deep learning). Windows has probably developed a very well trained network. And I guess Howdy uses one that is open source ?

> why doesn't just IR have the same accuracy as windows

The IR isn't a "intelligent" hardware, it just sends infrared rays which are reflects by your face in order to have more contrast with the surrounding grey. And thus be able to make a better image analysis.

boltgolt is surely much more knowledgeable than I am on this subject. But that's basically the ideas.

>  on Linux I get either fast at best and almost never accurate.

If you are really looking for security, I think it is best to have a strong password, even under Windows. I think facial recognition is more of a convenience than a security tool. The camera can be fool, but a password cannot. But this is just my personal opinion ! (not based on scientific research or something else)

kiraitachi, 2022-01-26:
Hi.

Im sort on the same boat. Although I can notice some differences. I have an Asus Zenbook but setting my device_path to /dev/video2 both white and red lights go on (cam and IR).

Issue is that when doing a "sudo howdy add" the facial recognition fails:

verage darkness: [72.72458], Threshold: 50.0
Face detection image too dark

Any tip?

EmixamPP, 2022-01-26:
In `sudo howdy config` change the value of  `dark_threshold` to 80 for example.

kiraitachi, 2022-02-18:
Yep thanks mate that fixed the issue!

> In `sudo howdy config` change the value of `dark_threshold` to 80 for example.

yi-ge, 2022-06-17:
Face detection image too dark during daylight,  use /dev/video0 during daylight and /dev/video2 at night?

kiraitachi, 2022-06-17:
Hi I think this issue can be closed. Who opened it didnt gave any further feedback and in my case it was fixed.

Thanks

closed 2022-06-17 by boltgolt