boltgolt/howdy #609: Unsupported Extension

dav-is, 2021-11-10:
Face recognition is working fine, but I see this warning logged each time

```
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
MFX: Unsupported extension: /dev/video2
```

----

[x] I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 35

Machine: Lenovo X1 Carbon, Gen 9

Howdy version (`sudo howdy version`): `2.6.1`

EmixamPP, 2021-11-12:
This driver is related to your Intel igpu. I guess, it is used for hardware acceleration. 

I don't think that the problem comes from Howdy. But I can't be wrong.

BaaptM, 2022-01-03:
Same issue on Fedora 35 with iHD driver. Face recognition works fine also but MFX keeps complaining. 
Found this quick workaround to quiet things down until I find a true fix : https://github.com/boltgolt/howdy/issues/595#issuecomment-964224376

closed 2026-11-16 by boltgolt