boltgolt/howdy #613: Import Error (x1 yoga gen6)

salvoc81, 2021-11-17:
An import error is raised when running `howdy test` or `add`  

```
sudo howdy test

Traceback (most recent call last):
  File "/usr/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 11, in <module>
    from recorders.video_capture import VideoCapture
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 6, in <module>
    import cv2
ImportError: libfmt.so.8: cannot open shared object file: No such file or directory
[salvocos@yogagnome ~]$ sudo howdy test
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 9, in <module>
    import cv2
ImportError: libfmt.so.8: cannot open shared object file: No such file or directory
```

I've searched for similar issues already, and my issue has not been reported yet.

System info:
```
inxi -b
System:    Kernel: 5.15.2-arch1-1 x86_64 bits: 64 Desktop: GNOME 41.1 Distro: ArcoLinux
Machine:   Type: Convertible System: LENOVO product: 20XYCTO1WW v: ThinkPad X1 Yoga Gen 6
CPU:       Info: Quad Core 11th Gen Intel Core i7-1165G7 [MT MCP]
Graphics:  Device-1: Intel TigerLake-LP GT2 [Iris Xe Graphics] driver: i915 v: kernel
           Device-2: Syntek Integrated Camera type: USB driver: uvcvideo
           Display: x11 server: X.Org 1.21.1.1 driver: loaded: modesetting unloaded: fbdev,vesa resolution: 3840x2400~60Hz
           OpenGL: renderer: Mesa Intel Xe Graphics (TGL GT2) v: 4.6 Mesa 21.2.5
```
Howdy 2.6.1

boltgolt, 2021-11-17:
You have a broken dlib installation, reinstalling dlib and python3-dlib might work?

salvoc81, 2021-11-18:
It seems to be a problem related to the AUR package...  
Let's hope it get fixed soon.

JPYamamoto, 2021-12-29:
Any updates on this?

EmixamPP, 2021-12-30:
`sudo pacman -S fmt` ? It is in the Extra repo: <https://archlinux.org/packages/extra/x86_64/fmt/>.

closed 2026-11-16 by boltgolt