boltgolt/howdy #616: Anyway to use libcamera instead of v4l?

k4ever, 2021-11-20:
I'm running Arch Linux on a Microsoft Surface Pro 4. The cameras on the Surface Pro 4 use libcamera instead of v4l. However, I was able to get Howdy to work with the Surface Pro 4 by creating a loopback device to /dev/video42, and setting this as the Howdy video device. The loopback was created using the Gstreamer Loopback Device for other Applications instructions on this website: https://github.com/linux-surface/linux-surface/wiki/Camera-Support

The problem is that this command needs to be ran to initial the camera prior to using Howdy:

gst-launch-1.0 libcamerasrc camera-name='\\_SB_.PCI0.I2C2.CAMF' ! \
    video/x-raw,width=1280,height=720,framerate=30/1,format=NV12 \
    ! videoconvert ! video/x-raw,format=YUY2 ! queue ! \
    v4l2sink device=/dev/video42

Is there a way to get Howdy to also use libcamera?

boltgolt, 2021-11-20:
Howdy does not currenty support that, but has a concept called "recorders". Every recorder is responsible for getting the video feed from the configured video device and delivering it to OpenCV. If you're interested i'm sure it's possible to write a recorder for your usecase and add it to v3.0.0

afalout, 2022-07-05:
How does the "recorder" concept work - does it depend on capturing video stream into a file and then processing it?
Could you please point me to the documentation/howto for it?
MS Surface Pro tablets will soon get a support for IR sensor so being able to use Howdy on Surface would be a dream come true...
Thanks!

Svampebob1, 2024-05-08:
Hi,
I'm also facing this issue.
Did you solve this?

afalout, 2024-05-08:
Dan Scally on Surface Linux camera project was working on libcamera support for Howdy, but that was maybe 6 months ago and then there was silence...
Needless to say, we wont have Howdy on any recent built-in cameras until there is libcamera support in Howdy. And/or gstreamer or pipewire, once they have libcamera support. OK so Im not being very exact but something like that. Lower level lib/HW support > Higher level lib abstraction layer > Applications. 
Until this is a reality, we need libcamera support in Howdy. Sigh.

Svampebob1, 2024-05-08:
> Dan Scally on Surface Linux camera project was working on libcamera support for Howdy, but that was maybe 6 months ago and then there was silence... Needless to say, we wont have Howdy on any recent [...]

I might have a workaround for this. I have made a systemd service  for this command: gst-launch-1.0 libcamerasrc camera-name='\SB.PCI0.I2C2.CAMF' !
video/x-raw,width=1280,height=720,framerate=30/1,format=NV12
! videoconvert ! video/x-raw,format=YUY2 ! queue !
v4l2sink device=/dev/video42

I just need to find a way to start the service through pam, before triggering Howdy.
The plan is to stop the service again after a successful login, so that the webcam isn't on constantly.

djrscally, 2024-05-09:
> Dan Scally on Surface Linux camera project was working on libcamera support for Howdy, but that was maybe 6 months ago and then there was silence...

Yeah sorry; I haven't had much spare time lately (my wife had our third child 6 and a half months ago; from one kid to two was a doddle but it turns out that going from two to three really kills your time). I am actively working on both sides of this problem though; extending libcamera to handle the IR cameras on IPU3 devices and adding a libcamera recorder to Howdy. I recognise you've been waiting a long time @afalout - I'll try to get everything working even if not upstreamable and share it.

> I might have a workaround for this. I have made a systemd service for this command: gst-launch-1.0 libcamerasrc camera-name='\SB.PCI0.I2C2.CAMF' ! [...]

This kind of thing should work in the interim as long as you don't mind not using the IR camera.

Svampebob1, 2024-05-09:
> Yeah sorry; I haven't had much spare time lately (my wife had our third child 6 and a half months ago; from one kid to two was a doddle but it turns out that going from two to three really kills [...]

I totally understand this. No worries. Take care of your kids and wife :-).

> This kind of thing should work in the interim as long as you don't mind not using the IR camera.

Thanks. I'll share the results later.

Svampebob1, 2024-05-10:
I've figured it out, but this is quick and dirty! Do this on your own risk! It might be smart to have root access while setting this up and testing it, so that you can disable Howdy if it fails. 

Feel free to ask if you have any questions.

1. Set up and make sure that libcamera with v4l2loopback works according to the Wiki from this tutorial: https://github.com/linux-surface/linux-surface/wiki/Camera-Support#gstreamer-loopback-device-for-other-applications
2. Set up loopback to video device: `sudo vi /etc/systemd/system/libcamera_loopback.service `
3. Paste the following:
```
[Unit]
Description=libcamera loopback

[Service]
Type=simple
User=
#Group=
ExecStart=/usr/bin/gst-launch-1.0 libcamerasrc camera-name='\\\_SB_.PCI0.I2C2.CAMF' ! video/x-raw,width=1280,height=720,framerate=30/1,format=NV12 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video42
Restart=on-failure

[Install]
WantedBy=multi-user.target
```
4. Create a policy kit exception so that it can be run by all users in the system:
	1. `sudo vi /etc/polkit-1/rules.d/libcamera.rules`
	2. Paste the following:
```
polkit.addRule(function(action, subject) {
    if (action.id == "org.freedesktop.systemd1.manage-units" &&
        action.lookup("unit") == "libcamera_loopback.service" &&
        (action.lookup("verb") == "start" || action.lookup("verb") == "stop" || action.lookup("verb") == "restart")) {
        return polkit.Result.YES;
    }
});
```
5. Test that the service works and that the webcam light turns on and off when the service starts and stops:
```
sudo systemctl daemon-reload
sudo systemctl restart polkit
sudo systemctl start libcamera_loopback.service
sudo systemctl status libcamera_loopback.service
sudo systemctl stop libcamera_loopback.service
```
6. Install Howdy according to the instructions here: [Linux_generelt](:/34a049d826394e58a2c70b00c0112cde)
7. Create scripts for starting and stopping:
	1. `sudo vi /bin/start-libcamera-loopback.sh`
	2. Paste the following, close and save:
```
#!/bin/bash
systemctl start libcamera_loopback.service
```
	3. `sudo vi /bin/stop-libcamera-loopback.sh`
	4. Paste the following, close and save:
```
#!/bin/bash
systemctl stop libcamera_loopback.service
```
	5. `sudo chmod +x /bin/*-libcamera-loopback.sh`
8. Set up PAM with the following:
	1. Place this at the top:
`auth    required    pam_exec.so seteuid /bin/start-libcamera-loopback.sh`
	2. Place this at the bottom:
`auth    optional     pam_exec.so seteuid /bin/stop-libcamera-loopback.sh`

djrscally, 2024-05-10:
Sweet - thanks for that. For what it's worth I tried to [hack the work I'd done so far into working](https://github.com/djrscally/howdy/tree/libcamera_reader) and I **think** it should...but I haven't had time to test it on my surface yet and likely won't now for at least another week (I'll be away at a conference all next week and probably busy prepping until then). The actual image capture through libcamera works fine, the bit I've been stuck on is convincing opencv to actually convert the image buffers that you receive. To work around that I've just set it to slice out the Y plane from the NV12 buffers that the IPU3 produces and treat it as a gray buffer...it ought to work but as I say I haven't actually managed to test it yet. 

If you wanted to test this before then (bearing in mind that I haven't...) you'd need to have built the python bindings in libcamera (`meson configure -D pycamera=enabled`, requires libpython3-dev pybind11-dev) and in config.ini set `recording_plugin=libcamera` and `libcamera_camera_id="your camera id here"`. The camera IDs you can find using `cam -l`:

```bash
$ cam -l
[2:51:00.296914792] [21301]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+139-66f0c2fe-dirty (2024-05-09T10:11:19+01:00)
Available cameras:
1: 'USB  Live camera: USB  Live cam' (\_SB_.PCI0.GP17.XHC0.RHUB.PRT3-3:1.0-0c45:6366)
2: Internal front camera (\_SB_.PCI0.GP17.XHC1.RHUB.PRT3-3:1.0-13d3:5458)
```

I'll get the conversion working once I'm back and then at least the non-IR camera would be available.

afalout, 2024-05-10:
Much appreciated, Dan.
I will wait for the IR camera as plain photo one is not usable for any form of authentication in real world.
Please keep us updated, your work is very important for a lot of people.
Cheers

afalout, 2024-05-29:
> Those are USB cameras on my PC...the IDs for the Surface cameras aren't quite so horrific. I'll get the conversion working once I'm back and then at least the non-IR camera would be available.

Hi Dan,

just a quick check if you had time to move this forward?

Thanks
Andrej

djrscally, 2024-06-11:
> > Those are USB cameras on my PC...the IDs for the Surface cameras aren't quite so horrific. I'll get the conversion working once I'm back and then at least the non-IR camera would be available.
> 
> Hi Dan,
> 
> just a quick check if you had time to move this forward?
> 
> Thanks Andrej

Yes, some progress - I can capture from the IR camera with the unpacking into a normal greyscale format done by libcamera. The other problem is that since we can't drive the IR LEDs the image is too dark; I have libcamera mostly correcting that now but it's not quite right. Once that's working I'll share it so you can try it out - it still won't be in a finished state until I have time to do a bunch of cleanup, but it should work at that point.

Thanks
Dan

sanicki, 2024-08-20:
👋 Another Microsoft Surface (Go 3) user following this work. Any updates?

djrscally, 2024-08-21:
> 👋 Another Microsoft Surface (Go 3) user following this work. Any updates?

None, but having been prompted by lots of people in the last week to work on this again I will try to drive it forwards

afalout, 2024-08-24:
> > 👋 Another Microsoft Surface (Go 3) user following this work. Any updates?
> 
> None, but having been prompted by lots of people in the last week to work on this again I will try to drive it forwards

Hi Dan @djrscally  - just noted this article that might be of interest?

https://jgrulich.cz/2024/08/19/making-pipewire-default-option-for-firefox-camera-handling/

Thank you for you efforts!
Andrej

djrscally, 2024-08-24:
> Hi Dan @djrscally - just noted this article that might be of interest?
>
> https://jgrulich.cz/2024/08/19/making-pipewire-default-option-for-firefox-camera-handling/
>
> Thank you for you efforts! Andrej

Heh good picture of Kieran there. The pipewire/WebRTC support is not of direct help to getting the IR camera working, but the Software ISP that's mentioned there is what's making the IR camera work too. The problem with how it's set up on the IPU3 (and why this has taken so long) is that the IR camera's data comes out raw (not through the hardware ISP) and packed into a format that's not supported anywhere - the Software ISP is relatively new and enables us to drive exposure automatically (without which it's useless) and also unpack it to a commonly supported format.

I think I just have one bug left; currently auto-exposure drives the exposure to max and leaves it there for some reason. Following that the libcamera reader I added in the link above needs to be updated to accept the Y10 format that the IR camera will write out, and then in theory it should work...

afalout, 2024-09-19:
Hi Dan - any luck with auto-exposure ??

Thanks!

heisenbergers, 2025-02-03:
> I've figured it out, but this is quick and dirty! Do this on your own risk! It might be smart to have root access while setting this up and testing it, so that you can disable Howdy if it fails.
> 
> Feel free to ask if you have any questions.
> 
> 1. Set up and make sure that libcamera with v4l2loopback works according to the Wiki from this tutorial: https://github.com/linux-surface/linux-surface/wiki/Camera-Support#gstreamer-loopback-device-for-other-applications
>    Certainly! Here’s the revised document with the markdown issues fixed:
> 2. Set up loopback to video device:
>    ```
>    sudo vi /etc/systemd/system/libcamera_loopback.service
>    ```
> 3. Paste the following:
>    ```
>    [Unit]
>    Description=libcamera loopback
>    
>    [Service]
>    Type=simple
>    User=
>    #Group=
>    ExecStart=/usr/bin/gst-launch-1.0 libcamerasrc camera-name='\\\_SB_.PCI0.I2C2.CAMF' ! video/x-raw,width=1280,height=720,framerate=30/1,format=NV12 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video42
>    Restart=on-failure
>    
>    [Install]
>    WantedBy=multi-user.target
>    ```
> 4. Create a policy kit exception so that it can be run by all users in the system:
>    
>    1. ```
>       sudo vi /etc/polkit-1/rules.d/libcamera.rules
>       ```
>    2. Paste the following:
>       ```
>       polkit.addRule(function(action, subject) {
>           if (action.id == "org.freedesktop.systemd1.manage-units" &&
>               action.lookup("unit") == "libcamera_loopback.service" &&
>               (action.lookup("verb") == "start" || action.lookup("verb") == "stop" || action.lookup("verb") == "restart")) {
>               return polkit.Result.YES;
>           }
>       });
>       ```
> 5. Test that the service works and that the webcam light turns on and off when the service starts and stops:
>    ```
>    sudo systemctl daemon-reload
>    sudo systemctl restart polkit
>    sudo systemctl start libcamera_loopback.service
>    sudo systemctl status libcamera_loopback.service
>    sudo systemctl stop libcamera_loopback.service
>    ```
> 6. Install Howdy according to the instructions.
> 7. Create scripts for starting and stopping:
>    
>    1. ```
>       sudo vi /bin/start-libcamera-loopback.sh
>       ```
>    2. Paste the following, close and save:
>       ```
>       #!/bin/bash
>       systemctl start libcamera_loopback.service
>       ```
>    3. ```
>       sudo vi /bin/stop-libcamera-loopback.sh
>       ```
>    4. Paste the following, close and save:
>       ```
>       #!/bin/bash
>       systemctl stop libcamera_loopback.service
>       ```
>    5. ```
>       sudo chmod +x /bin/*-libcamera-loopback.sh
>       ```
> 8. Set up PAM with the following:
>    
>    1. Place this at the top:
>       ```
>       auth    required    pam_exec.so seteuid /bin/start-libcamera-loopback.sh
>       ```
>    2. Place this at the bottom:
>       ```
>       auth    optional     pam_exec.so seteuid /bin/stop-libcamera-loopback.sh
>       ```

Hey, did you ever manage to get it to run on gnome on startup? I've tried several things, like changing the path directory or executing it after networking begins, but nothing really works to get something close to a native Windows Hello experience

afalout, 2025-03-04:
> > The problem with how it's set up on the IPU3 (and why this has taken so long)
> 
> Well, no, that's what made it **hard**. It took so long because I stopped working on it for various reasons.

Hello again Dan, just wondering if you had a chance to move this forward?

Much appreciated,
Andrej

producer456, 2026-09-08:
For anyone still on the v4l2loopback + `gst-launch` workaround: I've published a Howdy recorder that talks to libcamera directly, so there's no loopback device and nothing to start beforehand.

https://github.com/producer456/howdy-libcamera-yunet

It's a `recording_plugin = libcamera` option; you point `libcamera_camera` at a sensor model or a libcamera index. Needs `python3-libcamera`, no other new dependency.

Two things worth knowing that cost me time:

- **libcamera's own V4L2 compat shim doesn't solve this.** When several sensors sit behind one media device they advertise the same `/dev/videoN` nodes, so every open resolves to camera index 0 — usually the rear camera — and libcamera 0.7 has no environment variable to pick a different one. Talking to libcamera directly is what lets you name the sensor you want.
- **Leave `frame_width` / `frame_height` at `-1`.** libcamera *crops* rather than letterboxes when you ask for a different aspect ratio, so a 4:3 request against a 16:9 sensor cuts the sides off and can push a centred face out of frame entirely. It looks exactly like a broken face detector and is not one.

Developed and tested on Qualcomm `camss` (Snapdragon X, Ubuntu 26.04, aarch64) against Howdy 3.x, the meson build. The libcamera side is generic, but that's the only camera stack I've tested, so I'd be interested to hear whether it works on the Intel Surface models this thread started with.

The repo also contains an unrelated second piece (a YuNet detector option) which isn't what this issue is about, so I'll leave it there.

kbingham, 2026-09-08:
> Two things worth knowing that cost me time:
 
  * **libcamera's own V4L2 compat shim doesn't solve this.** When several sensors sit behind one media device they advertise the same `/dev/videoN` nodes, so every open resolves to camera index 0 — usually the rear camera — and libcamera 0.7 has no environment variable to pick a different one. Talking to libcamera directly is what lets you name the sensor you want.

This is an interesting one. I thought we'd designed the v4l2_compat layer to enumerate cameras as separate video devices.
It sounds like a bug to me. You could file a bug report at https://bugs.libcamera.org/

  * **Leave `frame_width` / `frame_height` at `-1`.** libcamera _crops_ rather than letterboxes when you ask for a different aspect atio, so a 4:3 request against a 16:9 sensor cuts the sides off and can push a centred face out of frame entirely. It looks exactly like a broken face detector and is not one.

This sounds like a bug too and I think I saw a fix for this one on the list recently.

https://patchwork.libcamera.org/patch/28178/

If you could test that patch - it would be helpful to reply to the patch with a "Tested-by:" tag if it works and solves your issue to help get it merged.

afalout, 2026-09-08:
> Developed and tested on Qualcomm `camss` (Snapdragon X, Ubuntu 26.04, aarch64) against Howdy 3.x, the meson build. The libcamera side is generic, but that's the only camera stack I've tested, so I'd be interested to hear whether it works on the Intel Surface models this thread started with.

 @producer456 - This is great news, your work is appreciated!

On the Surface side, looking at https://github.com/linux-surface/linux-surface/wiki/Camera-Support
OV7251 sensor that provides IR on majority of Surface devices is still not supported 

Except for Surface Laptop 3 (AMD) apparently.

@kbingham - any chance OV7251 support could be considered soon?

@djrscally at some point had a "tree with the mostly-working code" but then life interfered ...

Andrej