boltgolt/howdy #621: Installing howdy issue

tanujdargan, 2021-11-30:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 110, in <module>
    handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
  File "/usr/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3'
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kubuntu 21.10

Howdy version (`sudo howdy version`): howdy doesn't work but shows as it's installed

christianrauch, 2021-12-03:
I faced the same issue and in my case this was caused by having pip install via pip (I once did a `sudo pip3 install -U pip`).

To fix the issue, first verify that pip3 is installed via pip itself. In this case, a `which pip3` will give you `/usr/local/bin/pip3`. You then have the uninstall this "pip3 from pip" and install the Ubuntu pip3 package:
```
sudo pip3 uninstall pip
sudo apt install --reinstall python3-pip
```

If all goes well, `which pip3` should give you `/usr/bin/pip3` (i.e. not in local) and the installation script should find pip.

expl0r3rgu1, 2022-02-08:
> I faced the same issue and in my case this was caused by having pip install via pip (I once did a `sudo pip3 install -U pip`).
>
> To fix the issue, first verify that pip3 is installed via pip itself. [...]

This worked and a wrong installation of pip seems to be the actual problem. I saw a lot of issues similar to this one

closed 2022-12-30 by tanujdargan

dm30, 2023-04-20:
I tried to follow the steps above but I obtain the error "externally-managed-environment" when trying to uninstall. It happens too with other programs that interact the same way with python. 
Thank you for your help
 
[image]