boltgolt/howdy #622: Installation fails because of missing dependencies

LeGAlbert, 2021-12-02:
When trying to install Howdy with aur, when using makepkg -s this happens : 

```shell
==> Making package: howdy 2.6.1-2 (jeu. 02 déc. 2021 13:58:45)
==> Checking runtime dependencies...
==> Installing missing dependencies...
error: target not found: pam-python
warning: 'pam-python' is a file, did you mean -U/--upgrade instead of -S/--sync?
error: target not found: python-dlib
warning: 'python-dlib' is a file, did you mean -U/--upgrade instead of -S/--sync?
error: target not found: python-face_recognition
warning: 'python-face_recognition' is a file, did you mean -U/--upgrade instead of -S/--sync?
error: target not found: python-face_recognition_models
warning: 'python-face_recognition_models' is a file, did you mean -U/--upgrade instead of -S/--sync?
==> ERROR: 'pacman' failed to install missing dependencies.
==> Missing dependencies:
  -> pam-python
  -> python-dlib
  -> python-face_recognition
  -> python-face_recognition_models
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
```
When doint ```sudo pamac -install python-dlib``` my computer crashes ...
What am i supposed to do ?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Manjaro

Howdy version (`sudo howdy version`): 2.6.1-2

EmixamPP, 2021-12-04:
You should use the command `makepkg -csi` instead.

EmixamPP, 2021-12-04:
May be, try to use the GUI software installer ? Manjaro's is by far the best I've tested.

LeGAlbert, 2021-12-04:
Same happens with the gui :/
I started with the gui (cuz im à noob in Linux) and then tried with commands, ill try -csi when ill be back home!

Albert COSOREANU
Étudiant EPITECH Paris 1ère année
***@***.***
+33 7 86 93 95 69

EmixamPP, 2021-12-04:
Strange, because the GUI should solve this kind of problem.
The outputed errors by the GUI were the same ?

LeGAlbert, 2021-12-04:
When i use the gui the whole computer dies and i have to hard reboot it by pressing thé button

Albert COSOREANU
Étudiant EPITECH Paris 1ère année
***@***.***
+33 7 86 93 95 69

EmixamPP, 2021-12-04:
I guess you'r Manjaro installation is broken :cry: 
What is your kernel version ? `uname -r`
Try to update it using Manjaro settings manager <https://wiki.manjaro.org/index.php/Manjaro_Settings_Manager>. Install for example the version 5.15, reboot and select this version on grub.

closed 2026-11-16 by boltgolt