boltgolt/howdy #623: Does not work with GNOME GUI password popups

ExposedCat, 2021-12-11:
When I try to enter any application using the GNOME PolKit window (vscode, systemctl), the camera LED flashes, the popup doesn't appear until I re-enable howdy. `sudo -i` works fine with same setup and face position

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution:
Fedora 35, GNOME Shell 41.1
Howdy version:
Howdy 2.6.1

IlChitarrista, 2023-02-27:
Hello, I've been testing howdy out on Arch and I think I've got a nice setup.

So, the way we handle this stuff is through the config files in `/etc/pam.d`
When we first configure `howdy` it is necessary to add `auth sufficient pam_python.so /lib64/security/howdy/pam.py` to the sudo file to make it so that howdy is called when we use sudo.

In order to get Gnome GUI password popups to work with howdy we can add the same line to `/etc/pam.d/polkit-1` 
This is similar to how it works on gdm; In the ArchWiki there is also guidance, for GUI centric situations, to make the non face password input be usable too. This was only necessary for gdm on my system.

Hope this helps.

dukrous, 2025-02-13:
Just to add to this you need to make sure you have a `polkit-1` file already in `/etc/pam.d`. If you don't, copy it from `/usr/lib/pam.d/polkit-1`. After this, add the like @IlChitarrista suggests and it works right away. If you just create the file it won't authenticate correctly.

Tested this on Gnome 47 with Howdy 2.6.1 from Arch AUR package.

grohan47, 2025-02-27:
For Fedora, follow @dukrous instructions. Then, modify the copied file and add the following line to the top 
`auth       sufficient   pam_howdy.so`\

please do NOT follow @IlChitarrista instructions, because those only work on Arch.

grohan47, 2025-03-01:
> For Fedora, follow [@dukrous](https://github.com/dukrous) instructions. Then, modify the copied file and add the following line to the top `auth sufficient pam_howdy.so`
> 
> please do NOT follow [@IlChitarrista](https://github.com/IlChitarrista) instructions, because those only work on Arch.

@boltgolt, could we also include these polkit instructions in the setup guide in the howdy-beta COPR?

closed 2026-11-16 by boltgolt