boltgolt/howdy #624: Vertical side Feature request

neeks76, 2021-12-17:
Greeting to developer.

I'm very thankful to this solution.
It's good work in horizontal.

I would like to request a feature is vertical side.
When I make notebook to vertical from horizontal, then the howdy can't work.

I tried to take a picture face on horizontal, but this solution not expected.

Plz, solve or new feature release.

boltgolt, 2021-12-20:
Could be implemented quite easily, but would really slow down detection if enabled

matan-arnon, 2021-12-30:
Will be trying to implement this as my first open source contribution.

boltgolt, 2021-12-30:
That would be very appreciated! I was thinking of adding a config option that allows you to specify extra angles Howdy should run face detection on, like: 

```
rotation_angles = 0,90 
```

Which would just simply rotate every other frame 90 degrees

chenxiex, 2024-06-22:
I'm using howdy beta on Arch Linux with [https://aur.archlinux.org/packages/howdy-beta-git](https://aur.archlinux.org/packages/howdy-beta-git). The vertical side verification works with howdy pam, but howdy test doesn't show the rotated frames, nor does it show the green circle.

boltgolt, 2026-11-16:
Hey, thanks for requesting this! v3 adds a `rotate` option under `[video]`. Set it to `1` to check both landscape and portrait, or `2` for portrait only, and Howdy will recognise you with the laptop held vertically. Closing this as implemented!

closed 2026-11-16 by boltgolt